mik3y/usb-serial-for-android #256: Taking time to get response while sending byte data in soap form in telegisis  Silicon lab dongle

Shyamlaltatva, 2020-01-16:
When writing byte data that are in soap form connecting with telegisis  Silicon lab dongle with Cp21xxSerialDriver 

.  socket?.write(toByteArray) take time to response  and it give response when writing new data bytes

Will it take input in queue?  I want to get response without taking time for wait and not in next serial data write . Can it work asyn while writing data. Please suggest

kai-morich, 2020-01-16:
Write is is basically split into 2 phases:
a) from android to usb device
b) from usb device to serial port
If the send buffer on the usb device is not full, the write call returns after a). If the buffer is full, e.g. your write is larger or older data is still to be send, when enough data has been send over the serial port and the rest is in the buffer.
There is currently no method in this library to check that everything is fully written to the serial port. Typically this is handled by application specific protocol, e.g. using request-response pattern.
If you want to have it fully async, you should use separate thread + queuing in android app.

kai-morich, 2020-01-22:
moved answer to wiki: https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting/_edit#is-write-synchronous

closed 2020-01-22 by kai-morich

Shyamlaltatva, 2020-01-23:
Thanks for your quick reply. I have customized SerialInputOutputManger as per my requirement for mot losing data.