mik3y/usb-serial-for-android #257: Link to device_filter.xml on the main page is not working anymore

thirstyone, 2020-01-31:
2. Copy device_filter.xml to your project's res/xml/ directory.

kai-morich, 2020-01-31:
when I click on the link, the xml file content is shown. Please describe issue in detail

thirstyone, 2020-01-31:
oops! wrong repository (got lost in my two tabs between two browsers). please, disregard this issue report - it was intended for this project: 
https://github.com/eric2036/usb-serial-for-android
however, will your implementation support the stm32 usb-serial driver? do I need to add anything to device_filter.xml? (haven't tried it yet - building an stm device)

kai-morich, 2020-01-31:
STM devices use the CDC-ACM driver. There should be no need to modify the library. 

If your device uses VID/PID not known to the library, you should add a custom prober and if you want autostart you can also add the IDs to the xml copied into your app

closed 2020-02-06 by kai-morich