mik3y/usb-serial-for-android #258: Error opening device: init failed: #2

gokhanplt, 2020-02-03:
[image]

[image]

I want to connect one scale, but it gives such an error. Could you help?

kai-morich, 2020-02-03:
please try latest version with fix for ch340 initialization

gokhanplt, 2020-02-05:
> please try latest version with fix for ch340 initialization

[image]
The connection was successful. But it returns such a value.
It returns normal value in Windows.

kai-morich, 2020-02-05:
you very likely have to adapt the baud rate or other serial settings in the sample app

kai-morich, 2020-02-12:
fix included in v2.2.1

closed 2020-02-12 by kai-morich