mik3y/usb-serial-for-android #260: Unable to detect OTG storage in unity android App.

Divyanshu-del, 2020-02-13:
I know this question is a little off topic but I am struggling to communicate with the USB OTG through the android app. I have used an android java object to detect all storages using "GetExternalFilesDirs" but it only detects the internal storage and SD card storage. I have tried on stack overflow but no replies that can lead me in the right direction and unity answers is no help either. I was wondering if you could point me in the right direction. Here is my code that runs on Start() in unity.

                using (AndroidJavaClass unityPlayer = new AndroidJavaClass("com.unity3d.player.UnityPlayer"))
                {
                   using (AndroidJavaObject context = unityPlayer.GetStatic<AndroidJavaObject>("currentActivity"))
                   {
         // Get all available external file directories (emulated and USBdrives)
                     AndroidJavaObject[] externalFilesDirectories =  context.Call<AndroidJavaObject[]>("getExternalFilesDirs", (object)null);
                debugTExt.text = externalFilesDirectories.Length.ToString();
               DebugTExt2.text = externalFilesDirectories.Length.ToString();
               Debug.Log(externalFilesDirectories.Length.ToString());
                     AndroidJavaObject emulated = null;
                     AndroidJavaObject USBdrive = null;
                

                      for (int i = 0; i < externalFilesDirectories.Length; i++)
                          {
                	         AndroidJavaObject directory = externalFilesDirectories[i];
                    
                            using (AndroidJavaClass environment = new AndroidJavaClass("android.os.Environment"))
                            {         
                              bool isRemovable = environment.CallStatic<bool>("isExternalStorageRemovable", directory);
                              bool isEmulated = environment.CallStatic<bool>("isExternalStorageEmulated", directory);
                              if (isEmulated)
                              {
                                emulated = directory;
                              }
                              else if (isRemovable && isEmulated == false)
                              {
                                USBdrive = directory;
                              }
                             }
                          }

                // Return the USBdrive if available
                if (USBdrive != null)

                { 

                 //   USBdrive.Call<string>("getAbsolutePath");
                    if (File.Exists(USBdrive.Call<string>("getAbsolutePath") + "/serial.txt"))
                    {
                     /*   StreamReader extdata = File.OpenText(@USBdrive.Call<string>("getAbsolutePath") + "/serial.txt");
                        while(extdata.ReadLine()!=null)
                        {

                            textfileData = extdata.ReadLine().ToString();
                        }
                        debugTExt.text += " "+  textfileData;
                        DebugTExt2.text += " " + textfileData;
                        extdata.Close();
                        debugTExt.text += " "+  textfileData;
                        DebugTExt2.text += " " + textfileData;*/
                        textfileData = File.ReadAllText(USBdrive.Call<string>("getAbsolutePath") + "/serial.txt");
                        if (textfileData == SerialKeys.instance.keys[0])
                        {
                            keyDrive = USBdrive.Call<string>("getAbsolutePath");
                            filename = keyDrive + "/DeviceID.txt";
                        }
                        else textfileData = null;
                    }
                    else
                       
                        keyMissing.gameObject.SetActive(true);

                }
                        else
                //emulated.Call<string>("getAbsolutePath");
                keyMissing.gameObject.SetActive(true);

            }

      }
               
#endif`

``

How can I use your plugin to detect the USB storage at runtime and read a file inside it?

Please, Help me with this, I am struggling with this for a week now. I am trained in c# so I am unable to use java also. I will be so grateful if you can guide me.

kai-morich, 2020-02-13:
This library is only for USB serial devices.
USB mass storage devices are handled by Android and usually show up as additional folder returned by `getExternalFilesDirs`. I recommend to test with file manager Apps.

closed 2020-02-13 by kai-morich

Divyanshu-del, 2020-02-14:
I told you getExternalFilesDir is fetching sd card and internal storage but not usb storage

kai-morich, 2020-02-14:
have you checked if your Android device recognizes the USB storage and shows it in 'My Files' or other file manager app? Maybe you can change some Android settings to get it recognized or have to format it before. If it is recognized by Android you can access it with `getExternalFilesDir ` and do not have to reimplement an USB mass storage device driver.

Divyanshu-del, 2020-02-14:
Ok, I'll check but I am sure my device supports OTG. I watch movies by connecting it all the time and share media as well.

Divyanshu-del, 2020-02-20:
Sir, I checked but couldn't point out the fix. 
Please help me in fetching path of the connected usb in c# for android. GetExternalFilesDir is not working on 90% of devices.

eclectice, 2020-02-21:
Your Unity USB filesystem access problem has nothing to do with this
USB-Serial library. This library fetches data over serial connection using
USB cable interface similar to serial RS-232 connection.