mik3y/usb-serial-for-android #261: crass on iary data

8th-bit, 2020-02-13:
Hi,

I am using this lib to communicate with arduino nano. If nano sends string data with serial.print and it all works fine. If I use serial.write with nano to transmit binary data, it works well as long as any transmitted bytes does not have highest bit set. If highest bit (0x80) is set, library stops runing as run function is stoped by something. Both systems are using same settings, that is 8-bit data, one stop bit, no parity. 

Has anyone seen anything like this, what might be the problem? 

br Mikko

kai-morich, 2020-02-13:
Please provide more info. Without info one can only guess. Could be a conversion issue from signed Java byte to unsigned integer with wrong handling for negative byte with highest bit set.

8th-bit, 2020-02-16:
Thanks for your comment, I think I found the issue. Nothing to do with your lib, it was display lib I am using that did not support 8th bit.

closed 2020-02-16 by 8th-bit