mik3y/usb-serial-for-android #263: Read Data from OTG in unity

Divyanshu-del, 2020-02-20:
Dear Sir, 
I am a freelance Indie Developer.
I am struggling for more than a month to read USB devices (pen drive) in my unity android app. I am unable to do this with getExternalFilesDir. I am not at all experienced in Java. I need you to guide me in using this plugin in c#. Please help me, sir, I will even pay you if you take a fee but please help me with this little module. I have to get a connected USB path in my c# app.

It works in windows with DriveInfo,getDrives(). My whole project is complete.

Here is what I tried but failed drastically.  

``
using (AndroidJavaClass unityPlayer = new AndroidJavaClass("com.unity3d.player.UnityPlayer"))
                {
       
            using (AndroidJavaObject context = unityPlayer.GetStatic<AndroidJavaObject>("currentActivity"))
                   {
              
                // Get all available external file directories (emulated and USBdrives)
                AndroidJavaObject[] externalFilesDirectories =  context.Call<AndroidJavaObject[]>("getExternalFilesDirs", (object)null);

              
                AndroidJavaObject emulated = null;
                     AndroidJavaObject USBdrive = null;
                

                      for (int i = 0; i < externalFilesDirectories.Length; i++)
                          {
                	         AndroidJavaObject directory = externalFilesDirectories[i];
                    
                            using (AndroidJavaClass environment = new AndroidJavaClass("android.os.Environment"))
                            {         
                              bool isRemovable = environment.CallStatic<bool>("isExternalStorageRemovable", directory);
                              bool isEmulated = environment.CallStatic<bool>("isExternalStorageEmulated", directory);
                              if (isEmulated)
                              {
                                emulated = directory;
                              }
                              else if (isRemovable && isEmulated == false)
                              {
                                USBdrive = directory;
                              }
                             }
                          }

                // Return the USBdrive if available
                if (USBdrive != null)

                {
               
                    //   USBdrive.Call<string>("getAbsolutePath");

                    if (File.Exists(USBdrive.Call<string>("getAbsolutePath") + "/serial.txt"))
                    {
                       
                        textfileData = File.ReadAllText(USBdrive.Call<string>("getAbsolutePath") + "/serial.txt");
                        if (textfileData == SerialKeys.instance.keys[0])
                        {
                            keyDrive = USBdrive.Call<string>("getAbsolutePath");
                            filename = keyDrive + "/DeviceID.txt";
                        }
                        else textfileData = null;
                     
                    }
                    else
                       
                        keyMissing.gameObject.SetActive(true);
                    
                }

                        else
                //emulated.Call<string>("getAbsolutePath");
                keyMissing.gameObject.SetActive(true);
                
            }

      }
               
#endif
``
Please help me with this and I will be grateful to you for a lifetime.  

Thanks in advance.

eclectice, 2020-02-21:
It does not look like a USB-Serial use case. It is more like trying to
access a USB file system via OTG mode.

Divyanshu-del, 2020-02-21:
Okay. I am sorry yeah you got it right. It is OTG mode. How do I get my app to read the connected OTG?

kai-morich, 2020-02-22:
Please ask in Unity community

closed 2020-02-22 by kai-morich