mik3y/usb-serial-for-android #264: you can instantiate driver classes directly if you know what you're doing; just supply a compatible UsbDevice.

maxelrad, 2020-02-22:
I have successfully communicated with an Arduino Mega 2560 using the Simple USB Terminal project.
My personal capabilities fall short of specifying the driver directly in the  USB Serial code project which doesn't probe for it..
The device  id would be  CdcAcm  Vendor:2341  ProductId:0042
ATmega32U4 based USB
Direction on how to specify this directly would be very helpful to my specific needs and would improve my understanding of the library usage.

Thank you.
michael

kai-morich, 2020-02-22:
Insert your vi
Insert your VID,PID here https://github.com/kai-morich/SimpleUsbTerminal/blob/master/app/src/main/java/de/kai_morich/simple_usb_terminal/CustomProber.java

maxelrad, 2020-02-22:
Installed the class -changed vendor & product

Works fine!

Really appreciate your response and the projects.

closed 2020-02-23 by kai-morich