mik3y/usb-serial-for-android #267: Cp21xxSerialDriver: Read not working

charno, 2020-03-05:
Hi,

I'm trying to use usb-serial-for-android to download data from dive computers for the subsurface application (https://github.com/Subsurface-divelog/subsurface/). Now I have the problem that when I `call read(byte[1], 3000)` on Cp21xxSerialDriver (or FtdiSerialDriver) I don't get any data, the returnvalue of the function is always 0.

I have added come logging to the CommonUsbSerialPort-Class (see below). With this logging I get the following output: `bulkTransfer: ReadEndpoint=UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=64,mInterval=0], timeout=3000, nread=-1, readMax=1`
Since nread is -1 I recognize that the underlying android function has some problems.

When I use `timeout = 0` (thus async read) I get the same result of `nread=-1`.

Writing sems to work, at least I get the output `CommonUsbSerialPort: Wrote amt=2 attempted=2` which seems right.

I tried using the method `ioctl(ud->fd, USBDEVFS_RESET)` from https://github.com/mik3y/usb-serial-for-android/issues/262 to reset the device before connection, but to no avail.

If you want to have a look at the calling code, have a look at https://github.com/charno/subsurface/blob/android-serial/android-mobile/src/org/subsurfacedivelog/mobile/AndroidSerial.java. Relevant JNI code is at https://github.com/charno/subsurface/blob/android-serial/core/serial_usb_android.cpp.

Environment:
- Galaxy S10 with Android 10 (non-rooted)
- Application using QtMobile
- Configuration: baudrate=115200, databits=8, parity=2, stopbits=0
- timeout: 3000 (or 0 for testing, same result)

Thanks for your help!

Added logging:
```
@Override
    public int read(final byte[] dest, final int timeout) throws IOException {
        if(mConnection == null) {
            throw new IOException("Connection closed");
        }
        int nread;
        if (timeout != 0) {
            // bulkTransfer will cause data loss with short timeout + high baud rates + continuous transfer
            //   https://stackoverflow.com/questions/9108548/android-usb-host-bulktransfer-is-losing-data
            // but mConnection.requestWait(timeout) available since Android 8.0 es even worse,
            // as it crashes with short timeout, e.g.
            //   A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x276a in tid 29846 (pool-2-thread-1), pid 29618 (.usbserial.test)
            //     /system/lib64/libusbhost.so (usb_request_wait+192)
            //     /system/lib64/libandroid_runtime.so (android_hardware_UsbDeviceConnection_request_wait(_JNIEnv*, _jobject*, long)+84)
            // data loss / crashes were observed with timeout up to 200 msec

            int readMax = Math.min(dest.length, MAX_READ_SIZE);

            nread = mConnection.bulkTransfer(mReadEndpoint, dest, readMax, timeout);
            Log.d(TAG, "bulkTransfer: mReadEndpoint=" + mReadEndpoint + ", timeout=" + timeout + ", nread=" + nread + ", readMax=" + readMax);

        } else {
            Log.d(TAG, "endpoint: " + mReadEndpoint);

            final ByteBuffer buf = ByteBuffer.wrap(dest);
            if (!mUsbRequest.queue(buf, dest.length)) {
                throw new IOException("Queueing USB request failed");
            }
            final UsbRequest response = mConnection.requestWait();
            if (response == null) {
                throw new IOException("Waiting for USB request failed");
            }
            nread = buf.position();
            Log.d(TAG, "timeout=" + timeout + ", nread=" + nread);
        }
        if (nread > 0) {
            return readFilter(dest, nread);
        } else {
            return 0;
        }
    }
```

lsusb -v:
```
Bus 003 Device 017: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP2102/CP2109 UART Bridge Controller [CP210x family]
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2102 USB to UART Bridge Controller
  iSerial                 3 0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 CP2102 USB to UART Bridge Controller
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

charno, 2020-03-05:
I just tested the USB Serial Terminal application, and with this app it works fine... so the problem is probably in the calling code...

kai-morich, 2020-03-05:
please try https://github.com/kai-morich/SimpleUsbTerminal and check for differences to your code.

I have never used buffer size 1, maybe the USB stack gets confused when using buffers smaller than USB request size 64?

charno, 2020-03-05:
Thanks for your help, Kai. That was indeed the problem. I thought I tried bigger read sizes, but I used Math.min(readLen, 64) instead of Math.max(readLen, 64) by accident.

closed 2020-03-05 by kai-morich