mik3y/usb-serial-for-android #270: connection to cdc failed :android permission denied

jins6611, 2020-03-15:
I am getting this error when i have a serial and video interface. I am unable to connect

kai-morich, 2020-03-15:
probably your USB permission handling is missing. Please have a look at permission handling in https://github.com/kai-morich/SimpleUsbTerminal/blob/master/app/src/main/java/de/kai_morich/simple_usb_terminal/TerminalFragment.java

jins6611, 2020-03-15:
This issue is occurring for the same app in playstore too.

jins6611, 2020-03-16:
Issue is fixed i changed the target sdk to 27 and removed android.hardware.usb.host feature from manifest

closed 2020-03-16 by jins6611

kai-morich, 2020-03-16:
looks like android.hardware.usb.host is not really required, so I will try to remove from the example app.

https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal does not use android.hardware.usb.host in manifest, has target sdk 28 and minsdk 18, so looks very similar to your solution. Do you have any idea, why your change solved the issue.

jins6611, 2020-03-17:
> looks like android.hardware.usb.host is not really required, so I will try to remove from the example app.
>
> https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal does not [...]

 I think video and CDC interface both will try to use the usb host hardware feature if the mobile device support this