mik3y/usb-serial-for-android #271: read() gives empty but on data listener is working

JairajJangleSLIOT, 2020-04-02:
I am getting empty byteArray when I do serialPort.read(data, timeout) on button click but the read data listener is giving correct data. On the other end, I am using Arduino which is continuously sending data to the Android device. I want to capture what the Arduino is sending at that instance on button click and do not want to implement USB on data listener.

kai-morich, 2020-04-03:
this write error can have different reasons. typical are timeout too short, connection already closed

JairajJangleSLIOT, 2020-04-05:
If I do not use SerialInputOutputManager then I get this error while writing: `java.io.IOException: Error writing 6 bytes at offset 0 length=6 Android USB`. And direct read is not working even when I do that. 
I am not able to find any example app for direct R/W or am I missing something.

kai-morich, 2020-04-05:
[this](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/androidTest/java/com/hoho/android/usbserial/DeviceTest.java#L1329) test uses direct read and basically all of the tests use direct write

JairajJangleSLIOT, 2020-04-06:
Thanks for the solution, I will test this out and post the updates.

juCortez, 2020-04-08:
Hi, my name is Juliano. I am facing exactly same problem. Did you found a way to read data??

JairajJangleSLIOT, 2020-04-08:
@juCortez No, for direct reading I tried removing SerialInputOutputManager, which is causing a write error: `java.io.IOException: Error writing 6 bytes at offset 0 length=6 Android USB` where 6 is the number of bytes I am writing in one go; direct `serial.read(...)` is still not working. I have checked the test cases @kai-morich suggested to look at, I am still working on understanding it, will post updates here if I find anything new.

juCortez, 2020-04-08:
ok, thanks for reply!
best regards,

kai-morich, 2020-04-08:
please have a look at the latest example app, it supports event-based and direct read

JairajJangleSLIOT, 2020-04-10:
@kai-morich Thanks for updating the example app, direct Read works!!

closed 2020-04-10 by JairajJangleSLIOT

BhaurajBiradar001, 2023-05-24:
@JairajJangleSLIOT - What code you changed to work read directly?