mik3y/usb-serial-for-android #272: No provision to R/W unsigned byte array data

JairajJangleSLIOT, 2020-04-03:
The device which I am using is a motor driver which mandatorily required unsigned Byte Array data for it to operate. I tried converting the list of integer to unsigned byte array in Kotlin as well as Java but the write function of this library does not support unsigned byte array. Am I missing something? Is there any way to send an unsigned byte array via USB using this library? 

What I tried:
```
var msg = ubyteArrayOf(
        129.toUByte(),
        101.toUByte(),
        254.toUByte(),
        101.toUByte(),
        0.toUByte(),
        101.toUByte(),
        250.toUByte()
    )

// Wont work
// serialPort.write(msg)
```

kai-morich, 2020-04-03:
the library supports unsigned bytes as you can see in this test
https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/androidTest/java/com/hoho/android/usbserial/DeviceTest.java#L788
probably your conversion is wrong, but I am not a kotlin expert

JairajJangleSLIOT, 2020-04-03:
I will look into it and post the updates here. I will also try using Java.

JairajJangleSLIOT, 2020-04-10:
Update: I am finding it difficult to access hardware to test the app for unsigned bytes. I will post updates here as soon as I test it.

kai-morich, 2020-05-11:
closed due to inactivity

closed 2020-05-11 by kai-morich