mik3y/usb-serial-for-android #275: CDC driver mode is missing?

guiglass, 2020-04-28:
I just tested your app here https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=en_US

and it works allowing us to specify CDC driver type.

Then i installed the example apk on this Github and it keeps reporting "no driver"

Did I not install the correct example? Or do you not provide CDC for free?

How do I specify the driver type for this Github example to be CDC?

kai-morich, 2020-04-28:
you have to add your VID/CID combination here: https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/CustomProber.java

guiglass, 2020-04-28:
I found that info right after posting my question. We managed to get our device to communicate! This example is superb, thank you!

closed 2020-04-28 by kai-morich