mik3y/usb-serial-for-android #276: Received data is coming in as single bytes instead of a buffered sequence?

guiglass, 2020-04-29:
Hi, Thanks for this plugin! I have had success using the example with one of our USB to serial devices, but today I tried another device from a different manufacturer and I saw an unexpected result.

It appears the data was streaming in as individual bytes (this caused our app to process each byte individually and temporarily lock up).

I examined the same data using an unmodified usb-serial-for-android Github example, and I can see each byte is read individually and for some reason not being buffered properly:
https://i.imgur.com/YEIpElk.png

I then switched back to my other USB to serial device and sent the same data. And that came in just fine as a single buffered string (220 bytes long).

I also used the [serial terminal app](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=en_US) from the Play Store. Both devices appear to be buffered (220 bytes long).

Is there anything that might be causing this example to read each byte individually? Are there timers or buffer options I can change? Or maybe my devices are using different termination characters?

kai-morich, 2020-04-29:
this is normal behavior and depends on the USB device as each vendor uses different thresholds:
https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time

guiglass, 2020-04-29:
Thanks, I think I will have to create a handler/runnable to buffer data until the line goes silent for ~200ms indicating the end of scan. 

Scans range in duration but do not exceed 100ms.

I am not certain how the read timeout works in this library, but I am wondering if it would be possible to set the read timeout for ~200ms so it will buffer in the entire frame? Would using a read timeout block the app at all? Or am I misunderstanding its purpose?

Thanks again.!

kai-morich, 2020-04-29:
The USB device will send data wheneever its internal thresholds are reached. The read timeout only affects, on how long you the read command will block when no data is availble. 
You should also try to read continuosly, else data will be lost due to USB device internal receive buffer overflow.

closed 2020-05-14 by kai-morich

gvsrini, 2022-07-08:
_You should also try to read continuosly, else data will be lost due to USB device internal receive buffer overflow._

Can we have an example implementation of this. I am sure it will be of help to many users of the library

kai-morich, 2022-07-09:
For continuous reading you can use the `SerialInputOutputManager` class as shown on the project start page