mik3y/usb-serial-for-android #279: How Can reciever data in a service?

chenhua1008611, 2020-05-17:
this is my service code:
---------------------------------------------------------------------------------------
public class XXService extends Service implements SerialInputOutputManager.Listener{

    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {
        Log.d(LOG_TAG, "onStartCommand");
        deviceId = intent.getIntExtra("device",0);
        portNum = intent.getIntExtra("port",0);
        if(usbPermission == UsbPermission.Unknown || usbPermission == UsbPermission.Granted){
            connect();
        }
        return START_STICKY;
    }

  @Override
    public void onNewData(byte[] data) {
        receive(data);
    }

    @Override
    public void onRunError(Exception e) {
        status("connection lost: " + e.getMessage());
        disconnect();
    }

 private void receive(byte[] data) {
        SpannableStringBuilder spn = new SpannableStringBuilder();
        spn.append("receive " + data.length + " bytes\n");
        if(data.length > 0) {
            spn.append(parseAscii(data) + "\n");
        }
        String temp = parseAscii(data);
        if (temp.contains("XXXX")){
            //doChangeGPSWork
        }
    }
}
--------------------------------------------------------------------
how can solve this problems?

kai-morich, 2020-05-17:
your code fragment looks incomplete. you also have to open the socket and set the service as listener, e.g. as used [here](https://github.com/kai-morich/SimpleUsbTerminal/blob/master/app/src/main/java/de/kai_morich/simple_usb_terminal/TerminalFragment.java#L246).
You also have to consider that the data can be recieved in arbitrary sized [fragments](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time) and you have to buffer data in `recieve` until your message is complete.

chenhua1008611, 2020-05-24:
Can you leave a personal email? To further communicate and solve project problems?

kai-morich, 2020-05-24:
please try code from https://github.com/kai-morich/SimpleUsbTerminal