mik3y/usb-serial-for-android #283: Correct way to implement serial poller

dvergon, 2020-05-27:
Hi! I'm trying to use your library (pretty neat, thanks) to make a serial poller: ask for the status of the device on the other side of the serial port periodically.

Made a Thread whose job is to send a poll stream every time it can

private class Poller implements Runnable{

        @Override
        public void run(){

            while(!Thread.interrupted()){

                if(pollStartAllowed){

                    byte[] poll = new byte[1];
                    poll[0] = intToByte(83);

                    byte[] formattedStream = formatStream(poll, false);

                    try {
                        serialWrite(formattedStream);
                    } catch (IOException e) {
                        e.printStackTrace();
                    }

                    pollStartAllowed = false;
                }
            }
        }
    }

The main thread (Activity) has the SerialInputOutputManager.Listener interface implemented, with which I receive the data using onNewData().

Both, read and write wait times, are set to 200ms.

The problem is:
When the project worked (now it doesn't for an unknown reason) it was able to poll (write) no problem, but, when trying to read the response, it could handle just a second or second and a half worth of data, then, it stopped reading and kept polling.

My guess is that the HW buffer is full or the baud rate was to high or the timeouts too small or a combination of all of those three.

Is there any chance that you can check the implementation over GitHub?
I'll leave the link right here: https://github.com/dvergon/SerialTest

Thanks for all your work and help.

dvergon, 2020-05-27:
[image]
Made some progress.

Config
9600 baud
no parity
1 stop bit
300 write timeout
700 read timeout

In the image you can see that I send two bytes (127 80) and the emulator responds with 10 bytes.
The first two times works like a charm, that's what I expected, but, after the second transmission, I receive the expected 10 bytes, plus two times the last byte of the last stream. After that, the software (Java/Android) stops receiving the data, even though the emulator keeps responding to the poll.

So:
- The software (Java/Android) keeps polling perfectly
- After a few (in this case, two) received streams, it stops receiving data
- The emulator keeps responding to the poll

Is there something I'm missing? Maybe clearing the buffer?

I did clear the buffer with the provided method, but it doesn't work or I'm using it wrong:

this.currentConnection.purgeHwBuffers(false, true); 

Hope you can enlighten me.

Thanks!

kai-morich, 2020-05-30:
maybe your serial device needs some time for processing and you need a delay before sending the next request or your purge deletes some valid data, that is still in send/receive buffer

kai-morich, 2020-06-20:
closed inactive issue

closed 2020-06-20 by kai-morich