mik3y/usb-serial-for-android #284: sendAcmControlMessage() result ignored?

4ntoine, 2020-05-28:
Hey, I can see in the [source code](https://github.com/mik3y/usb-serial-for-android/blob/33149b66fd4fbfb313cb5d88ec1936936a543049/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriver.java#L300) `sendAcmControlMessage()` call result is ignored.

In my fork i get negative result time to time (and i don't have such devices to test with) and i stop working with device and consider initialization failed.

Do you think it's ok to ignore the result?

kai-morich, 2020-05-29:
I haven't seen such issues with my CDC devices. Maybe you use some unsupported baud rate or other values?

Regarding your previous comment on the return value:
so far all methods using sendAcmControlMessage are only sending data so there is no relevant result length in the return value. If there would be methods reading data, they would have to check the return value.

4ntoine, 2020-05-29:
@kai-morich Thanks for replying.

Nope, exactly the same code works (no baud rate there (just `mConnection.controlTransfer(0x21, 0x22, 0x00, mInterfaceNum /* actually 0 */, null, 0, 0);`)) in different ways returning negative values on some devices.
I can'y see any correlation with manufacturer, device, Android OS version but i have the logs (Firebase Crashlytics with events and logs and stacktrace)

kai-morich, 2020-06-19:
closed as not reproducible

closed 2020-06-19 by kai-morich