mik3y/usb-serial-for-android #286: Private class FtdiSerialPort to protected

uholeschak, 2020-06-07:
Hello,
could you please change the class FtdiSerialPort to protected:
protected FtdiSerialPort(UsbDevice device, int portNumber) {
Otherwise the new Visual Studio bindings generator is not exporting the class any more.

By the way, for the usbSerialSample I have to add in build.gradle:
defaultConfig {
        vectorDrawables.useSupportLibrary = true
    }
otherwise I get an error during build.

Regards,
Ulrich

uholeschak, 2020-06-11:
Thanks, but you did not add the
vectorDrawables.useSupportLibrary = true
in build.gradle ...

kai-morich, 2020-06-11:
on my devices worked with out this flag, but as I have it in my other apps, I also added it here with 3e7fd9a74892c76e077ab4ec77e855f871005f79