mik3y/usb-serial-for-android #288: Compatible with PIC 16F886?

shamoilarsi, 2020-06-21:
Hey, I'm trying to communicate to a PIC 16F886 from an android app using this library. I'm currently not able to get any success. I was just curious if it is even possible to communicate directly with the IC? 

I can change the IC's program as per my requirements. 

Also, if this specific PIC doesn't work, is there any other what would?
If not, is there any IC that we could add in between to transmit data?

kai-morich, 2020-06-21:
this app can directly communicate with microcontrollers that include an USB port using the CDC/ACM serial port protocol. Your PIC doesnt look like it has USB, so you have to add the usual FTDI, CP2012, CH340, ... modules to connect the serial lines

shamoilarsi, 2020-06-21:
Hey, thank you for your reply.
The PIC 16F886 has UART support. But even then does communication using DMX signals (IC 75176) which transforms differential signals into digital I/Os any use? 

Can we replicate the handshake those modules provide with the help of any IC?

kai-morich, 2020-06-22:
you have to convert UART to USB with one of FTDI, CP2102, CH340, PL2303, ...

shamoilarsi, 2020-06-24:
Hey, okay thank for the reply!

closed 2020-06-24 by shamoilarsi