mik3y/usb-serial-for-android #290: Initial Bytes missing when receiving any data

UnisonControls, 2020-06-29:
Hi,

We are using your sample code for testing.

We make some modifications there.

We are able to transmitte 15 to 20 bytes using send function from TerminalFragment Function.

We want to read data when receiving something from PL2303 hardware.

We checked our hardware by connecting it to PUTTY software and able to get all data correctly.

So basically hardware working ok.

But whenever we check same thing on app we miss initial 2 to 3 bytes of data.

We go through to your Issue cases but not found any useful data.

Please give us correct way for solution.

Waiting for your quick reply.

Thanks and Best Regards,
Jehan Patel

kai-morich, 2020-06-29:
I havent seen such issue before. Maybe it's a power supply issue. The PL2303 is powered by the USB port from your PC or Android device. Does your serial device use the 3.3V or 5V from the PL2303 board?

UnisonControls, 2020-06-29:
Hi,

We have connected TX, RX, Gnd Pin from PL2303 to MCU.

Please find attached image for more details.

This connections work perfectly for PC communication via putty software.

[image]

kai-morich, 2020-06-29:
you should try lower baud rate or different uart device

UnisonControls, 2020-06-29:
I think Baud rate value is not problem. Because we transmission works perfect.

We already check with 9600 bps baud rate but problem remain same.

For any baud rate values we got correct result on PUTTY.

UnisonControls, 2020-06-30:
hi,
Is there any update on current conversation?
If you have any query then please tell us.
Waiting for your prompt response.

kai-morich, 2020-06-30:
please try with [Serial USB Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal) app

UnisonControls, 2020-07-01:
hi,
In your suggested app we got correct data.
might be we have some coding mistake in java file call.
we are beginner for this line, can you please check it once?
provide us your mail id so we can send you java file for this case.
waiting for your quick reply.

kai-morich, 2020-07-01:
please start with usbSerialExamples app in this project or https://github.com/kai-morich/SimpleUsbTerminal (which is a stripped down version of Serial USB Terminal). Then only change step by step and verify that it is still working.

kai-morich, 2020-07-23:
closed after 3 weeks inactivity

closed 2020-07-23 by kai-morich