mik3y/usb-serial-for-android #291: Would you support the newly discovered Ch340?

howtoplay, 2020-07-07:
That is, I have connected to a CH340, and then found a new CH340. I can directly connect to the new CH340 without disconnecting the original one.

kai-morich, 2020-07-07:
do you want to reconnect to a disconnected and reconnected CH340 or do you want to connect to multiple CH340 connected to an USB hub?
Both is possible if you loop over all devices/drivers returned by UsbSerialProber.getDefaultProber().findAllDrivers()

howtoplay, 2020-07-08:
UsbSerialProber.getDefaultProber().findAllDrivers() will return all new drivers, my original driver will be invalid. I wanna get a new driver when the devices found a new CH340, and my original driver still alive
---- Sorry, my English is not very good

kai-morich, 2020-07-08:
findAllDrivers returns new `UsbSerialDriver` and new `UsbDevice` objects. To identify and exclude your already open device you should use `usbSerialPort.getDevice().equal(other-device)` which compares the device file name. The device file name like _/dev/bus/usb/001/002_ is stable while a device is connected.

closed 2020-08-07 by kai-morich