mik3y/usb-serial-for-android #292: port.close() close also UsbDeviceConnection()

AsladaGSX, 2020-07-15:
My device has CDC DAT, COMM and CCID interface.
CommonUsbSerialPort.close() method call CommonUsbSerialPort.closeInt() method and UsbDeviceConnection.close().
CommonUsbSerialPort.closeInt() closes interfaces only, this is OK.
but calling UsbDeviceConnection.close() disconnect also CCID interface using by my application, this cause a problem in my application.

I think closing UsbDeviceConnection is my application's responsibility.

kai-morich, 2020-07-16:
you're right. The short example in [README](https://github.com/mik3y/usb-serial-for-android/blob/master/README.md) also shows that the deviceConnection is opened before and is just handed over to `UsbSerialPort.open()`, so it would make sense to close it separately after `UsbSerialPort.close()`.

But this would be an incompatible change, and it's also more convenient for the usual case to close the connection.

Instead I could think of an additional method `close(boolean closeUsbDeviceConnection)` or you pass in a derived `UsbDeviceConnection` object, where the `close()` method simply does nothing

kai-morich, 2020-07-18:
please try with individual `usbManager.openDevice(...)` for each interface. 
In my tests, it worked with individual `UsbDeviceConnection` for each port of a multi-port serial device and failed after first close if the connection was shared.

AsladaGSX, 2020-07-20:
It works. Thanks many!

closed 2020-07-20 by AsladaGSX

kai-morich, 2020-07-21:
updated javadoc comment with https://github.com/mik3y/usb-serial-for-android/commit/963729924b9fa70baf68da00ba384f8985f292eb