mik3y/usb-serial-for-android #294: cannot resolve symbol 'BuildConfig'

vijay180396, 2020-07-20:
i have downloaded the usb-serial-for-android(master) branch.And i import in android studio 4.0.1 .
it shows an error in **terminalfragment.java** at line of 47 **buildconfig** it not resolved.

vijay180396, 2020-07-20:
now i find out why i get this error.
Due to i not added dependencies of com.github.mik3y:usb-serial-for-android:2.3.0 in build.gradle(usbSerialExample).
but i cannot able to find **APPLICATION_ID ** in BuildConfig

kai-morich, 2020-07-20:
BuildConfig is generated when you build the project

vijay180396, 2020-07-21:
[image]

vijay180396, 2020-07-21:
and also it's doesn't assign any APPLICATIONID in **BuildConfig.class**
[image]

kai-morich, 2020-07-21:
APPLICATION_ID is only set in BuildConfig of the example app, not BuildConfig of the library, but you can also replace the variable with the actual value in your fork

[image]

vijay180396, 2020-07-21:
thank you sir!, now the build get works

closed 2020-07-21 by vijay180396

Jagdish-958, 2023-11-07:
not generated BuildConfig in java(generated)