mik3y/usb-serial-for-android #295: I want to donate your project.

sevketk, 2020-07-22:
first of all , thank you for your efforts. But, can you tell me how can I include your library locally in my project? which folder should i copy to my project and what settings should i make?

kai-morich, 2020-07-23:
You don't have to copy the library locally to your project.
You can automagically download it from jitpack as depicted [here](https://github.com/mik3y/usb-serial-for-android#quick-start).
If you really want to have it locally in your project, copy the `usbSerialForAndroid` folder and use in your app like [here](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialExamples/build.gradle#L29) and [here](https://github.com/mik3y/usb-serial-for-android/blob/master/settings.gradle#L1).

sevketk, 2020-07-23:
Actually, let me tell you my problem clearly. I developed some application on QT (C ++). And I move this software to the Android platform. qt does not implement url libraries, only implements local libraries (i think). QT  If you make one QT (android) example for serialport, I would like to donate $ 100 for the project.

sevketk, 2020-07-23:
I have run my project, everything except serial port, on android using Qt.

kai-morich, 2020-07-23:
if Android Qt can import JAR files, you can try [this](https://github.com/mik3y/usb-serial-for-android/wiki/Build-Variants#jar)

sevketk, 2020-07-23:
hi kai
I managed to add the library to my project. and I created my java file like this.

```
package org.qtproject.example;

import android.app.AlertDialog;
import android.content.Context;
import android.content.DialogInterface;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbManager;
import android.os.Bundle;
import androidx.fragment.app.Fragment;
import androidx.fragment.app.ListFragment;

import com.hoho.android.usbserial.driver.UsbSerialDriver;
import com.hoho.android.usbserial.driver.UsbSerialProber;

import java.util.ArrayList;
import java.util.Locale;

public class JNITest {

    // this method will be called from C/C++
        public static int fibonacci(int n)
        {
            UsbManager manager = (UsbManager) getSystemService(Context.USB_SERVICE);

            
        }

}
```

but when i run the project i get an error like this

`> Task :compileDebugJavaWithJavac FAILED
E:\kutuphane\belgelerim\QT Projects\JniTest2\android-build\src\org\qtproject\example\JNITest.java:26: error: cannot find symbol
            UsbManager manager = (UsbManager) getSystemService(Context.USB_SERVICE);
                                              ^
  symbol:   method getSystemService(String)
  location: class JNITest
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUİLD FAILED in 7s
12 actionable tasks: 2 executed, 10 up-to-date
Building the android package failed!
  -- For more information, run this command with --verbose.
22:07:20: The process "C:\Qt\5.15.0\android\bin\androiddeployqt.exe" exited with code 14.
Error while building/deploying project JniTest2 (kit: Android for armeabi-v7a,arm64-v8a,x86,x86_64 (Clang Qt 5.15.0 for Android))
When executing step "Build Android APK"`

it says to me 'getSystemService' not found. if i can, i will put the sample qt in your repostirities.

sevketk, 2020-07-23:
attrackt attention , 
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

kai-morich, 2020-07-24:
please check Android documentation for `getSystemService`

sevketk, 2020-07-24:
As you can see in my code, I imported the   import android.content.Context;  library.  Shouldn't it get to getSystemService method? I looked where you said.  What else should I do?

kai-morich, 2020-07-24:
getSystemService is a member of Context class, so you very likely first have to get a Context object

sevketk, 2020-07-29:
for example as AppCompatActivity ;) ok I understand.I think I gonna use 'extends Application' for my class. I hope works.

kai-morich, 2020-09-06:
closed due to inactivity

closed 2020-09-06 by kai-morich