mik3y/usb-serial-for-android #298: Some question about reading bytes

mucolores, 2020-08-05:
First of all, Sorry for bothering developers,
I've got some problem while developing android applications while using the library.

I'm reading a FTDI device's data, but I've cannot smoothly read the data bytes.
If I read 100 byte, there'll be some pairs of hex data 0x01 0x60 inside, which overwritten on the data from my device.
If there're 2 pairs of 0x01 0x60 data (which spacing 4 byte). The read function returns 96 correspondingly.
I've test the some device on PC but I didn't see the 0x01 0x60 data appears.
I've wondering is there any possible reason cause for the appears of 0x01 0x60 in my data. or did the bytes 0x01 0x60 stands with any special meaning?

My data be like
97 C7 7E 06 CE
97 D4 7F 16 E0
57 C9 01 60 7E // where the strange data appears
B6 CD 57 C9 7E
B6 CD 47 CA 7E
....
47 C9 7E 96 01 // another strange data appears
60 CE 97 CB 7E

I'm sure about that 0x01 0x60 is not part of my expecting data.
And it only appears when I called read function second time or further times.

kai-morich, 2020-08-05:
- maybe your baud rate is slightly wrong
- do you continously read all available data? if not, data might be overwritten in the FTDI device when it's internal buffer is full

mucolores, 2020-08-05:
Firstly, thanks you for answering my issue.
For the buad rate setting, I've follow the device document's buad rate specification, which is 921600
following on this viewpoint, I'm connecting the device via a USB to micro-usb converter, I'm wondering is there any possibility the converter will change the buad rate?
For reading the data continuously problem, I'll put my code bellow.

`usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        if(usbManager == null) {
            TestTextView.setText("No manager");
            return;
        }
        availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(usbManager);
        if (availableDrivers.isEmpty()) {
            String NVD = "No available drivers";
            TestTextView.setText(NVD);
            return;
        }
        // Open a connection to the first available driver.
        usbSerialDriver = availableDrivers.get(0);
        usbDeviceConnection = usbManager.openDevice(usbSerialDriver.getDevice());
        if (usbDeviceConnection == null) {
            // add UsbManager.requestPermission(driver.getDevice(), ..) handling here
            String NVC = "No available Connection";
            TestTextView.setText(NVC);
            return;
        }
        usbSerialPort = usbSerialDriver.getPorts().get(0); // Most devices have just one port (port 0)
        try {
            usbSerialPort.open(usbDeviceConnection);
            usbSerialPort.setParameters(BUAD_RATE, UsbSerialPort.DATABITS_8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);
            String ACString = "AC";
            usbSerialPort.write(ACString.getBytes(StandardCharsets.UTF_8),0);
            byte[] tmp = new byte[102];
            int length = 0;
            length = usbSerialPort.read(tmp,WAIT_TIMEOUT);
            String tmpString = length + "\n" + ByteToHexString(tmp,5,2);
            ResultTextView.setText(tmpString);

            Start = System.currentTimeMillis();
            for(int i=0;i<2;i++)
            {
                Arrays.fill(tmp,((byte)0x00));
                tmp = new byte[4095];
                length = usbSerialPort.read(tmp,WAIT_TIMEOUT);
                byte[] newTmp = new byte[length];
                int LingCounter = 0;
                for(int j=0;j<length-1;j++)
                {
                    if(tmp[j] == (byte)0x01 && tmp[j+1] == (byte)0x60)
                    {
                        LingCounter++;
                    }
                }
                tmpString = "0x0160 : " + LingCounter + "\n" + length + "\n"; //+ ByteToHexString(tmp,5);
                ResultTextView.append(tmpString);
            }
            End = System.currentTimeMillis();
            String TimeString = "Time elapsed : " + (End-Start) + "\n";
            TestTextView.setText(TimeString);
        } catch (IOException ioe)
        {
            Toast.makeText(MainActivity.this, ioe.toString(), Toast.LENGTH_SHORT).show();
        }`

I'm reading the data continuously in the for loop and only but I still get the wrong data.

mucolores, 2020-08-06:
By the way,
I've tried your example APP, and check the input data.
I didn't see the error data like 0x01 0x60.
Maybe there's something wrong in my code.

May I ask for the method how you continuously reading data?

kai-morich, 2020-08-06:
please look at *event driven read* in [README.md](https://github.com/mik3y/usb-serial-for-android#quick-start)

mucolores, 2020-08-06:
Excuse me, I still got some problem with the _event driven read_ README.md
Will the onNewData function only be called once?
Or it will be called for every time which the buffer detect new data.

After I try to implement the _event driven read_ , the function onNewData seems only to be called once.

kai-morich, 2020-08-07:
You should not use timeout 0 when writing.
onNewData is called for every new snippet of recieved data.
Recieved data can be split into arbitrary packets (for both event driven and direct read) . So there might be a single byte packet with your 0x01 byte and the 0x60 byte might be in the next packet

mucolores, 2020-08-10:
I've found out that if I want to display the data on UI in function ```onNewData```, I'll need to wrapped with runonUIThread.
And the problem seems to be solved.

Thanks a lot.👍  :)

closed 2020-08-12 by kai-morich