mik3y/usb-serial-for-android #300: Event driven reading loads of data

rnd-ash, 2020-08-11:
Hi,

I am writing a project that has an arduino connected to an android headunit via USB, and the Arduino relays CAN Frames as 12 byte structs to the headunit over USB Serial (up to 300 per second).

I am using the event driven reading to relay data to a JNI method that unpacks the sent data into structs:
```kotlin
class SerialManager() : SerialInputOutputManager.Listener {

    override fun onNewData(data: ByteArray?) {
        // Add to our native buffer
        data?.let { CarComm.nativeCanbus.sendBytesToBuffer(it) }
    }

    override fun onRunError(e: Exception?) {
    }
}
```
I have noticed though, that as the Arduino is constantly sending data, it appears the Event driven reader actually misses data, which then means that data being sent is mis-aligned, causing loads of incorrect parsing of the sent data.

Is there any way to create a JNI or custom method that can buffer all the read data to a large buffer? - Or increase the frequency of polling for new data? - I have managed to do this using JVM using strings and decoding the data into bytes, but this proved to be way too slow. I would like to simply transmit structs over serial and re-pack them on the JNI side of things if possible.

The full Android code can be found [here](https://github.com/rnd-ash/MBUX-Port/tree/master/app/src/main)

kai-morich, 2020-08-11:
Hi

you could try moving your data processing to a different thread with data queue to not block the USB retrieve thread unnecessarily.
But you might also be run into this 'data loss' issue http://kai-morich.de/android/usb_troubleshooting.html

Kai Morich

rnd-ash, 2020-08-11:
That is what is already happening. Once the data is buffered in JNI, it is then read by another thread in JNI environment. The data loss is occurring during the buffering phase, not the parsing.

Is there a way I can reset the arduino just before polling for data? So The read thread and Arduino are synced. I think this is DTR if im not mistaken?

kai-morich, 2020-08-11:
Hi

yes, some sort of flow control will help. But you should take into account that a stop signal might not be effective immediately as some data might already be buffered in the usb2serial chip.
If you can modify the Arduino code, using some acknowledge from Android side after each packet might be more reliable.

Kai

kai-morich, 2020-08-12:
You could try with higher thread priority as mentioned [here](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#i-observed-read-data-loss-with-continuous-transfer-at-high-baud-rates)

rnd-ash, 2020-08-12:
OK that kind of helped! 

Issue now is that there is a huge delay with DTR and RTS enabled, as it appears SerialInputOutputManager is waiting for the full 4KB buffer to fill up...

Is there any way to adjust the buffer size to something much much smaller like 48 bytes?

kai-morich, 2020-08-12:
The buffering strategy is usb device specific. Typically a combination of buffer fill ratio and serial idle time. For FTDI you can control the latter with setLatencyTimer()

rnd-ash, 2020-08-13:
I fixed my issue.

Ended up creating a custom SerialInputOutputManager:
```kotlin
open class IOManager(readBufSize: Int, private val serialPort: UsbSerialPort, listener: Listener) : SerialInputOutputManager(serialPort, listener) {
    enum class ManagerState {
        STOPPED,
        RUNNING,
        STOPPING
    }
    private var state: ManagerState = ManagerState.STOPPED
    private val readBuffer = ByteBuffer.allocate(readBufSize)
    private val writeBuffer = ByteBuffer.allocate(4096)
    override fun getState(): State {
        return when(state) {
            ManagerState.STOPPED -> State.STOPPED
            ManagerState.RUNNING -> State.RUNNING
            ManagerState.STOPPING -> State.STOPPING
        }
    }

    override fun run() {
        synchronized(this) {
            check(state == ManagerState.STOPPED) { "Already running" }
            state = ManagerState.RUNNING
        }

        Log.i("IOManager", "Running ...")
        try {
            while (true) {
                if (state != ManagerState.RUNNING) {
                    Log.i(
                        "IOManager",
                        "Stopping mState=$state"
                    )
                    break
                }
                step()
            }
        } catch (e: Exception) {
            Log.w(
                "IOManager",
                "Run ending due to exception: " + e.message,
                e
            )
            val listener = listener
            listener?.onRunError(e)
        } finally {
            synchronized(this) {
                state = ManagerState.STOPPED
                Log.i("IOManager", "Stopped")
            }
        }
    }

    fun step() {
        // Handle incoming data.

        // Handle incoming data.
        var len = serialPort.read(readBuffer.array(), 100)
        if (len > 0) {
            val listener = listener
            if (listener != null) {
                val data = ByteArray(len)
                readBuffer[data, 0, len]
                listener.onNewData(data)
            }
            readBuffer.clear()
        }

        // Handle outgoing data.

        // Handle outgoing data.
        var outBuff: ByteArray? = null
        synchronized(writeBuffer) {
            len = writeBuffer.position()
            if (len > 0) {
                outBuff = ByteArray(len)
                writeBuffer.rewind()
                writeBuffer[outBuff, 0, len]
                writeBuffer.clear()
            }
        }
        if (outBuff != null) {
            serialPort.write(outBuff, 100)
        }
    }
}
```
This custom manager I found works best with a 64 byte buffer, not a 4096 one.

And using a Listener:
```kotlin
@ExperimentalStdlibApi
class SerialManager() : SerialInputOutputManager.Listener {
    override fun onNewData(data: ByteArray?) {
        // Add to our native buffer
        data?.let {
            CarComm.nativeCanbus.sendBytesToBuffer(it, it.size)
        }
    }

    override fun onRunError(e: Exception?) {
    }
}
```

Then the data is buffered within JNI and a separate p_thread thread does the processing work.
With this method, I'm able to run a 115200/s baud rate and loose 0 bytes of data over a full 3 hours of this running.

Would it be possible to open the read and write buffer sizes to avoid having to create a custom class to do that?

kai-morich, 2020-08-14:
Good point, I will make if configurable when back from vacation. 

Now I remember that I also use a derived class with custom buffer size in my 'Serial USB Terminal' app because at low baud rate with permanent data transfer, some USB devices send data as soon as available but others prefer to fill the full buffer causing a lagged UI.

rnd-ash, 2020-08-19:
I found out via tracing that my read method is causing extensively high CPU usage. It appears this performance impact is occurring when I read something and it times out due to no data.

Is there a way we can see how many bytes are available to read (Without reading)? - Similar to an Arduino's `Serial.avaliable()` function?

kai-morich, 2020-08-19:
Timeout 100msec is to short. You should use a signifficantly longer timeout or even better 0 for infinite timeout

kai-morich, 2020-09-12:
with v3.1.0 you can set buffer size at `SerialInputOutputManager`

closed 2020-09-12 by kai-morich