mik3y/usb-serial-for-android #302: When reading data, data is blocked or lost

howtoplay, 2020-08-13:
When I write a ch340 instruction to receive infrared data, the reading thread will block the data or missing data , whether readTimeout is set to 0 or not

kai-morich, 2020-08-13:
Maybe you have send wrong instruction. Check baud rate, hardware, wires, ... or try with serial terminal app

howtoplay, 2020-08-14:
I haven't send the wrong command, but when I changed the source code to the following snippet, there were no more errors

`private static final int MAX_QUEUE_READ_SIZE = 32;`

`ByteBuffer byteBuffer = ByteBuffer.allocate(MAX_QUEUE_READ_SIZE);
boolean queue;
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
	queue = mUsbRequest.queue(byteBuffer);
} else {
	queue = mUsbRequest.queue(byteBuffer, MAX_QUEUE_READ_SIZE);
}
if (!queue) {
	throw new IOException("Queueing USB request failed");
}
final UsbRequest response = mConnection.requestWait();
if (response == null) {
	throw new IOException("Waiting for USB request failed");
}
nread = byteBuffer.position();
byteBuffer.flip();
byteBuffer.get(dest, 0, nread);`

kai-morich, 2020-08-14:
Your snippet looks very similar to 
https://github.com/mik3y/usb-serial-for-android/blob/73d669c4dc9af4bccdfdd35b1ff4d8e6c9c16ce9/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CommonUsbSerialPort.java#L133 with timeout 0 but smaller buffer size than used by `SerialInputOutputManager`.

Could you try with a smaller buffer in `SerialInputOutputManager`? You might have similar queuing issues as discussed here #300

kai-morich, 2020-09-12:
with v3.1.0 you can set the buffer size at `SerialInputOutputManager`

closed 2020-09-12 by kai-morich