mik3y/usb-serial-for-android #303: Detect disconnect

HarrysLapTimer, 2020-08-13:
When calling read like
```
byte[] buf = new byte[len];
bytesRead = mPort.read(buf, 10000);
```
the call to read is non-blocking. Instead, if no data is available, zero bytes are returned. Is this expected behavior?

Furthermore, when I disconnect the USB port, read() continues to return 0 bytes instead of an IOException to signal the end of stream. Again, is this expected behavior? And if yes, how am I supposed to recognize a disconnect (except using the USB_DEVICE_DETACHED intent-filter)?

Thanks for this library, helped me a lot!

kai-morich, 2020-08-13:
Read should block for 10 seconds if no data is available. What kind of usb device do you use?

kai-morich, 2020-08-13:
Different low level usb calls are used for 
- read with timeout
- read w/o timeout

The non-timout variant returns error on disconnect.

The timeout variant internally has the same error code for timeout and real error. I thought about adding some time dependent logic to interpret error as real error if it is returned long before the timeout.

HarrysLapTimer, 2020-08-13:
I'll try the version without timeout. The driver used is FtdiSerialDriver, the USB device is a wired OBD II adapter.

kai-morich, 2020-08-13:
Ftdi devices return a status packet each 16ms, so the timeout for this device type is basically not configurable. The library could be enhanced to wait for the remaining time.

The non-timeout variant is best used with SerialInputOutputManager thread to avoid blocking your UI thread.

HarrysLapTimer, 2020-08-13:
Using no timeout (actually 0), read calls do not block either, but I get an IOException when disconnecting the device.

```
0818406099 00 --> call to SerialForAndroidTalker::read (., ., len: 256)
0818406099 01   --> starting blocking read...

java.io.IOException: Queueing USB request failed
     at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:153)
     at com.harrys.gpslibrary.sensors.SerialForAndroidTalker.read(USBSensor.java:798)
     at com.harrys.gpslibrary.sensors.UsbInputStream.read(USBSensor.java:205)
     at com.harrys.gpslibrary.sensors.StreamSensor$InputStreamingTask.doInBackground(StreamSensor.java:229)
     at com.harrys.gpslibrary.sensors.StreamSensor$InputStreamingTask.doInBackground(StreamSensor.java:149)
     at android.os.AsyncTask$3.call(AsyncTask.java:378)
     at java.util.concurrent.FutureTask.run(FutureTask.java:266)
     at java.lang.Thread.run(Thread.java:919)

0818406099 00 --> SerialForAndroidTalker::read () returns throwing an exception
```
The lines with a timestamp in the beginning is logging from my own code

HarrysLapTimer, 2020-08-13:
> Ftdi devices return a status packet each 16ms, so the timeout for this device type is basically not configurable. The library could be enhanced to wait for the remaining time.
> 

O.k, understood.

> The non-timeout variant is best used with SerialInputOutputManager thread to avoid blocking your UI thread.

Can't use the SerialInputOutputManager because my sensor framework is doing the stream handling. But reads are performed on a separate thread, so this should be as designed.

kai-morich, 2020-08-13:
The blocking read is also terminates by the status packet.

HarrysLapTimer, 2020-08-13:
> The blocking read is also terminated by the status packet.

Yes, understood. Would be great to have this filtered some day, but it is not a problem if one knows about it. ;-) With the exception thrown (zero timeout) it is fine for me now, thanks for the support!

closed 2020-08-13 by HarrysLapTimer

kai-morich, 2020-09-12:
both issues are fixed with v3.1.0
* added disconnect detection 
* no more early return for FTDI