mik3y/usb-serial-for-android #308: Prolific PL2303 does not work with 74880 Bd

hmueller01, 2020-09-17:
I can not communicate with my ESP8266 using 74880 Baud and the Prolific PL2303HX (Product: 2303, Vendor: 067b).
The received data is unreadable. It looks like the baud rate is not set correctly.
Using another USB serial converter (CP2102 or CH340) the baud rate 74880 works with usb-serial-for-android examples as well as with Serial USB Terminal (1.38). Using the Prolific on a MacBook it is working too (installed this [driver](http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41)). So it looks like the special baud rate is not set correctly.

kai-morich, 2020-09-18:
non-standard baud rates for PL2303 are currently not supported by this library.
Prolific does not publish the formula for non-standard baud rates, but their [web site](https://prolificusa.com/product/pl2303hx-rev-d-usb-serial-uart-bridge-controller/) refers to the linux kernel, where the formula is implemented and very likely contributed by prolific. The formula is available in [linux](https://github.com/torvalds/linux/blob/10b82d5176488acee2820e5a2cf0f2ec5c3488b6/drivers/usb/serial/pl2303.c#L516) and [freebsd](https://github.com/freebsd/freebsd/blob/e48e67dde88022d9f4352c9685cfa6855b7d9ba1/sys/dev/usb/serial/uplcom.c#L758).

hmueller01, 2020-09-18:
Ok, I added/ported the Linux code and look, it works. :-)
I can do a pull request. But I am not sure what needs to be commented when this code is copied from the Linux kernel. This part of the kernel is GPL-2.0 and your code is MIT ...

kai-morich, 2020-09-19:
I recently have rewritten the FTDI driver to change the license of this library from LGPL to MIT. 
To my understanding the license of this library would have to be changed to GPL if any GPLed code is copied into this libarary.

[Here](https://lore.kernel.org/patchwork/patch/599496/) the current formula and code was added to Linux.
Not sure if the function in the comment is really GPL licensable, as it actually is created be Prolific but not published anywhere.
The code is also very short and a straight forward implementation of that formula, not sure if such short code snippets are licensable at all.
Maybe we can ask the author of that Linux patch, if he is open to other licenses?

kai-morich, 2020-09-29:
implemented with 732e138630fddd22e61ef72bc180789c08ae3a12

closed 2020-09-29 by kai-morich