mik3y/usb-serial-for-android #311: NullPointerException: rhs == null

janimm, 2020-09-21:
3.1.0 causes NullPointerException: rhs == null on connect. I have been reported that connection did work on 3.0.1 library.
Device to connect uses VID: 0x483, PID 0x5740.

kai-morich, 2020-09-21:
please provide call stack

janimm, 2020-09-22:
Apparently the USB device in this case is a STM32F405RGT6 Microcontroller. This report has come from an user of our app, and unfortunately our debug logging system does not take full call stack. I do not have that device for myself for testing.

kai-morich, 2020-09-22:
there is no variable `rhs` in this library. the issue very likely is some layers above or below

janimm, 2020-09-23:
It could be the lower level. We were able to fix this by selecting the driver manually for this VID PID combo.

closed 2020-09-23 by janimm