mik3y/usb-serial-for-android #314: Get rid of permission dialog appearing again on restart of device

sweety-wo, 2020-09-26:
I am implementing this library on rooted device. There are many USB devices attached to device. I am recognizing the Scanner device by Type and allowing user to scan the data. First time it asked me for the permission and it worked well. But when I had restarted the device it again asked me for permission as I added that permission on Service level too.

kai-morich, 2020-09-26:
Android resets all permissions on reboot. I don't know if that can be bypassed on a rooted device.

sweety-wo, 2020-09-26:
Do you have any idea how can I access the device without asking for the permission? I mean how to bypass the permission stage and directly access the USB device?

kai-morich, 2020-09-26:
I am not aware of easy solutions. On a rooted device you might be able to write a C library to directly access the linux kernel functions or /dev/usb... devices.