mik3y/usb-serial-for-android #318: support flow control

kai-morich, 2020-10-12:
the library does not support flow control with RTS/CTS or DTR/DSR control lines or XON/XOFF characters.
it only supports set/get of flow control lines.

subvertallchris, 2020-12-02:
Hey Kai, could you elaborate on this? I see references to RTS, CTS, DTR, and DSR throughout https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java.

Am I misunderstanding what’s happening in that code? I’m far from an expert at this. :-)

kai-morich, 2020-12-02:
with the existing code you can manually set/reset/get the control lines, but it does not support automatic flow control, which would set/reset the control lines automatically when buffers are full

luke-lombardi, 2020-12-03:
Hey @kai-morich - I'm curious what you think it would take to add support for automatic flow control. Could you maybe point me to where you would have to set/get the RTS/CTS lines? I'm guessing wherever you call read, you'd have check how much room is left in the RX buffer and then set CTS appropriately - and a similar check for RTS when you call write. Does that make sense? Thanks

kai-morich, 2020-12-03:
@luke-lombardi - what you have sketched is manual flow control, which might work in some cases but typically is not sufficient, as the buffer in the device might already be full before you notice in your app. Instead automatic flow control has to be used, that is basically supported by all devices support and can be enabled with some special control commands, that are more or less documented.

TRtomasz, 2020-12-23:
@kai-morich 
something like this should be ok
although i don't know how to set XON and XOFF characters(check screenshot) maybe it can be done by manipulating `SET_FLOW_REQUEST`
```
private static int FLOW_CONTROL_DISABLED = 0x00;
private static int SET_FLOW_REQUEST = 2;
            public override void SetFlowControlOff()
            {

                int result = mConnection.ControlTransfer((UsbAddressing)REQTYPE_HOST_TO_DEVICE, SET_FLOW_REQUEST,
                        FLOW_CONTROL_DISABLED, mPortNumber + 1, null, 0, USB_WRITE_TIMEOUT_MILLIS);
                if (result != 0)
                {
                    throw new IOException("Set flow TR failed: result=" + result);
                }
            }
```
[image]

tunjid, 2021-03-18:
@kai-morich I was hoping to use this library with software flow control. Do you have any pointers or resources I can use? I'm not sure where the read buffer is in `CommonUsbSerialPort` that would signal that a XOFF should be sent, much less an XON.

CaseyBorders, 2022-07-27:
I was just wondering if you had an ETA for 3.5.0 with automatic flow control?

mparkison, 2024-02-09:
Another bump to see if automatic flow control is still being considered. This library is head and shoulders above the others I've tried (great work!), but that one requirement is forcing me to use a less robust option.

kai-morich, 2024-06-04:
Core functionality is relatively simple and nearly done, but integrating nicely in API and sample application show some areas that need general improvement. This will be delivered before, like timeout improvement in v.3.7.3 to be able to handle write() being blocked by remote side.

kai-morich, 2024-07-05:
88ca3f57c423941b4a918e91a7231449512fa1e0

closed 2024-07-05 by kai-morich