mik3y/usb-serial-for-android #320: Can't write more than 200 bytes one time

huijimuhe, 2020-10-15:
great projects!
but here are some issues i met:

1.if write more than 200 bytes once, there will occasional throw an exception ;if write more than 250 bytes once, there will often throw an exception ;if write more than 400 bytes once, there will always throw an exception.(CommonUsbSerialPort.java;line.213)

2.if you have a GPS module. you can't receive all the data. it's about 760 bytes a pack one time, and send every 1s. How can i fix that?

kai-morich, 2020-10-15:
Did you specify a to short timeout, so write can't finish in time?
Read data fragmentation is normal, see wiki.

huijimuhe, 2020-10-16:
forget check timeout parameter, i'll give it a shot, thanks!

closed 2020-10-19 by huijimuhe