mik3y/usb-serial-for-android #322: No control endpoint

rdonik, 2020-11-07:
Does anyone know why this issues is happing? 

<usb-device vendor-id="2414" product-id="1539"  />

Error happed in port.open(connection) code 

java.lang.RuntimeException: Unable to start activity ComponentInfo{novalabs.fisacalmodel/novalabs.fisacalmodel.MainActivity}: java.io.IOException: No control endpoint
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3271)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3410)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2017)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7397)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
     Caused by: java.io.IOException: No control endpoint
        at com.hoho.android.usbserial.driver.CdcAcmSerialDriver$CdcAcmSerialPort.openSingleInterface(CdcAcmSerialDriver.java:129)
        at com.hoho.android.usbserial.driver.CdcAcmSerialDriver$CdcAcmSerialPort.openInt(CdcAcmSerialDriver.java:101)
        at com.hoho.android.usbserial.driver.CommonUsbSerialPort.open(CommonUsbSerialPort.java:98)
        at novalabs.fisacalmodel.MainActivity.init(MainActivity.kt:102)
        at novalabs.fisacalmodel.MainActivity.onCreate(MainActivity.kt:24)
        at android.app.Activity.performCreate(Activity.java:7802)
        at android.app.Activity.performCreate(Activity.java:7791)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1307)

MyCode:
val manager = getSystemService(Context.USB_SERVICE) as UsbManager
val actionString: String = application.packageName.toString() + ".action.USB_PERMISSION"

 val customTable = ProbeTable()
        customTable.addProduct(0x96E, 0x603, CdcAcmSerialDriver::class.java)
        val prober = UsbSerialProber(customTable)
        val drivers = prober.findAllDrivers(manager)
        if (drivers.isEmpty()) {
            return
        }
        // Open a connection to the first available driver.
        val driver = drivers[0]
        val connection = manager.openDevice(device)
        if (connection == null) {
            val mPermissionIntent = PendingIntent.getBroadcast(this, 0, Intent(actionString), 0)
            manager.requestPermission(driver.device, mPermissionIntent)
        }
        val port = driver.ports[0] // // Most devices have just one port (port 0)
        port.open(connection)

kai-morich, 2020-11-07:
maybe it's not a CDC device?

rdonik, 2020-11-07:
@kai-morich if it is not CDC device, i was not able to get information about it, is it? 
UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=2414,mProductId=1539,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=FeiTian Ltd,mProductName=JavaCard Token V1.0,mVersion=1.06,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@29c693d,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=50,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=11,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=64,mInterval=0]]]]

kai-morich, 2020-11-07:
please ask the vendor

kai-morich, 2020-11-12:
basically all USB devices are enumerated and you can get information about them (except a few devices like mouse and keyboard that are already consumed by Android)

CDC devices offer certain endpoints. But as you got `No control endpoint` error, it very likely is no CDC device, but uses it's own protocol. You should ask the vendor about the device specific protocol

closed 2020-11-12 by kai-morich