mik3y/usb-serial-for-android #325: I want to register a multi port listener

pegaskim, 2020-11-23:
I want to register each listener for multi cdc port, but the following error occurs.
I don't know how to register listener for each port.
Below is the implementation code and error message where the current error occurs.
**For reference, for single port, it works without problem.**

    private final SerialInputOutputManager.Listener mListenerAVR =
            new SerialInputOutputManager.Listener() {
                @Override
                public void onNewData(byte[] data) {
                    if (debug) {
                        StringBuilder sb = new StringBuilder();
                        for (byte b : data) {
                            sb.append(String.format("%02X ", b));
                        }
                        Log.d(TAG, "ListenrAVR : Len = " + data.length + ", " + sb.toString());
                    }
                }

                @Override
                public void onRunError(Exception e) {
                    Log.d(TAG, "AVR Runner Stopped.");
                }
            };
    private final SerialInputOutputManager.Listener mListenerScanner =
            new SerialInputOutputManager.Listener() {
                @Override
                public void onNewData(byte[] data) {

                }

                @Override
                public void onRunError(Exception e) {
                    Log.d(TAG, "Scanner Runner Stopped.");
                }
            };

// permission allow and port open success...
// below code is register rx listener..
       //ThreadPoolExecutor executor = (ThreadPoolExecutor) Executors.newFixedThreadPool(2);
        if (mSerialPort[USBIndex_AVR] != null) {
            mUsbIoManager[USBIndex_AVR] = new SerialInputOutputManager(mSerialPort[USBIndex_AVR], mListenerAVR);
            Executors.newSingleThreadExecutor().submit(mUsbIoManager[USBIndex_AVR]);
//            executor.submit(mUsbIoManager[USBIndex_AVR]);
        } else {
            Log.e(TAG, "UsbSerialPort(AVR) is NULL");
        }

        if (mSerialPort[USBIndex_SCANNER] != null) {
            mUsbIoManager[USBIndex_SCANNER] = new SerialInputOutputManager(mSerialPort[USBIndex_SCANNER], mListenerScanner);
            Executors.newSingleThreadExecutor().submit(mUsbIoManager[USBIndex_SCANNER]);
//            executor.submit(mUsbIoManager[USBIndex_SCANNER]);
        } else {
            Log.e(TAG, "UsbSerialPort(SCANNER) is NULL");
        }
//        executor.shutdown();

//error message
2020-11-23 15:32:11.616 31820-1511/com.gen2wave.smartsledtestapp E/martsledtestap: JNI ERROR (app bug): attempt to use stale Global 0x284a (should be 0x2842)
2020-11-23 15:32:11.617 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: java_vm_ext.cc:577] JNI DETECTED ERROR IN APPLICATION: use of deleted global reference 0x284a
2020-11-23 15:32:11.617 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: java_vm_ext.cc:577]     from android.hardware.usb.UsbRequest android.hardware.usb.UsbDeviceConnection.native_request_wait(long)
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] Runtime aborting...
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] Dumping all threads without mutator lock held
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] All threads:
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] DALVIK THREADS (19):
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] "pool-3-thread-1" prio=5 tid=19 Runnable
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | group="" sCount=0 dsCount=0 flags=0 obj=0x171cf4f8 self=0x7889944800
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | sysTid=1511 nice=0 cgrp=default sched=0/0 handle=0x78867a8d50
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | state=R schedstat=( 10671271 762963 22 ) utm=0 stm=1 core=6 HZ=100
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | stack=0x78866a6000-0x78866a8000 stackSize=1039KB
2020-11-23 15:32:11.757 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | held mutexes= "abort lock" "mutator lock"(shared held)
[... 229 lines cut ...]
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] 
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642] "Binder:31820_1" prio=5 tid=13 Native
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | group="" sCount=1 dsCount=0 flags=1 obj=0x16f40020 self=0x7892531000
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | sysTid=1460 nice=0 cgrp=default sched=0/0 handle=0x788e7c3d50
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | state=S schedstat=( 462039 349000 8 ) utm=0 stm=0 core=5 HZ=100
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | stack=0x788e6cd000-0x788e6cf000 stackSize=991KB
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | held mutexes=
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   kernel: (couldn't read /proc/self/task/1460/stack)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #00 pc 00000000000cefa4  /apex/com.android.runtime/lib64/bionic/libc.so (__ioctl+4)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #01 pc 000000000008bde4  /apex/com.android.runtime/lib64/bionic/libc.so (ioctl+132)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #02 pc 0000000000058904  /system/lib64/libbinder.so (android::IPCThreadState::talkWithDriver(bool)+260)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #03 pc 0000000000058ae0  /system/lib64/libbinder.so (android::IPCThreadState::getAndExecuteCommand()+24)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #04 pc 00000000000592ac  /system/lib64/libbinder.so (android::IPCThreadState::joinThreadPool(bool)+60)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #05 pc 000000000007fac0  /system/lib64/libbinder.so (android::PoolThread::threadLoop()+24)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | group="" sCount=1 dsCount=0 flags=1 obj=0x16f80020 self=0x788dca9c00
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   kernel: (couldn't read /proc/self/task/1463/stack)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #05 pc 0000000000085330  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #02 pc 0000000000017c78  /system/lib64/libutils.so (android::Looper::pollOnce(int, int*, int*, void**)+56)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:209)
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   | state=R schedstat=( 71277076 762963 58 ) utm=4 stm=2 core=6 HZ=100
2020-11-23 15:32:11.758 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #09 pc 0000000000306b84  /system/framework/arm64/boot-framework.oat (art_jni_trampoline+148)
2020-11-23 15:32:11.759 31820-1511/com.gen2wave.smartsledtestapp A/martsledtestap: runtime.cc:642]   native: #43 pc 00000000000ea9e4  /apex/com.android.runtime/javalib/core-oj.jar (java.lang.Thread.run+8)
    
    --------- beginning of crash
2020-11-23 15:32:11.759 31820-1511/com.gen2wave.smartsledtestapp A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 1511 (pool-3-thread-1), pid 31820 (martsledtestapp)

kai-morich, 2020-11-23:
multi port listener work here: https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/androidTest/java/com/hoho/android/usbserial/CrossoverTest.java#L113. please try with two distinct connections to same device

kai-morich, 2020-12-15:
closed due to inactivity

closed 2020-12-15 by kai-morich