mik3y/usb-serial-for-android #327: Wrong data - send and receive to/from Arduino uno from example code

moorthi07, 2020-11-29:
When I send ,
1
Arduino returns and the 'receive' function output this result,
receive 5 bytes 
06 98 78 1E F8
--- (this changes into two outputs like 4bytes, 1 byte or 3 bytes , 2byte as if it is two read from the Arduino board)

My Arduino code is simple ,

void loop(){   
//forword();
if(Serial.available() > 0){  //if some date is sent, reads it and saves in state     
state = Serial.read();  
Serial.print(state); 

Thanks in advance.

kai-morich, 2020-11-30:
please check your baud rate

moorthi07, 2020-11-30:
Thanks for your response. 

Its the same in the example: 

`private int baudRate = 19200;`

moorthi07, 2020-12-04:
Works. Changed Serial.begin(19200) in Arduino code.

closed 2020-12-04 by moorthi07