mik3y/usb-serial-for-android #329: Is onNewData method run in a same thread?

DrakeXiang, 2020-12-07:
I'm creating a react-native app and not familiar with java or android native development, I'm using `onNewData` method to read data from device, there's a situation that I need to communicate with device many times in sequence, so I think it's better to loop the call syncronously, I set a flag field in the class that implements `SerialInputOutputManager.Listener` interface, and change it in `onNewData` method, so I thought I could check the flag to determine weather to proceed next loop call, but the flag seems not updated.

Here's the sample code:
```java
// USBDATAHandler
public class USBDataHandler implements SerialInputOutputManager.Listener {
    private static USBDataHandler instance = null;

    public boolean isDone = false;
    public byte[] parsedBytes = null;

    public static USBDataHandler create() {
        if (instance == null) {
            instance = new USBDataHandler();
        }
        return instance;
    }

    @Override
    public void onNewData(byte[] data) {
        try {
                readBuffer = data;
         
                byte[] res = parseResp(readBuffer);
                parsedBytes = res;

                if (res != null) {
                    toggleIsDone(true);

                    if (callBack != null) {
                        callBack.finish(res);
                    }
                } else {
                    rejectPromise("Read fail");
                }
        } catch (Exception e) {
            rejectPromise(e.getMessage());
        }
    }

```

```java
// MyModule
...
...
    private SerialInputOutputManager usbIoManager;

    private USBDataHandler usbDataHandler = USBDataHandler.create();

     usbIoManager = new SerialInputOutputManager(port, this.usbDataHandler);
     handlerExecutor = Executors.newSingleThreadExecutor();
     handlerExecutor.submit(usbIoManager);

    public void loopCall() {
         int count = 0;
         while (count < 100) {
               byte[] res = sendData();
         }
    }
...
...

   public void sendData() {
        try {
            this.usbDataHandler.resetWithoutFile(); // this reset the fields
            port.write(frameBytes, WRITE_WAIT_MILLIS);
            int count = 0;

            while (true) {
                Thread.sleep(1000);
                count++;

                if (this.usbDataHandler.isDone || count > 9) {
                    // This parsedBytes is null, why?
                    return this.usbDataHandler.parsedBytes;
                }
            }
        } catch (Exception ex) {
            throw ex;
        }

    }
```

I hope it's enough to present my intention, now the `while(true)` clause always get `null` but I'm certain that 9 seconds is long enough to read data, am I blocking the same thread that `onNewData` runs in or something?

kai-morich, 2020-12-07:
`onNewData()` is called in the background thread created by `Executors.newSingleThreadExecutor`. From the code snippet it's not obvious why it blocks

DrakeXiang, 2020-12-08:
@kai-morich Thanks for answering, I'll keep trying and close issue for now

closed 2020-12-08 by DrakeXiang