mik3y/usb-serial-for-android #331: mConnection.requestWait() hang up when data length==32

yaphone, 2020-12-10:
Hi,
I used usb-serial-for-android lib with a ch340 hardware, it recieve data from ch340, when the data length is not 32 or 32 * n, it works well, but it DOS NOT work when the data length is 32 or 32*n like "hellohellohellohellohellohellohe" (length = 32) or "hellohellohellohellohellohellohehellohellohellohellohellohellohe"(length = 64) . I debug it, found it block at `mConnection.requestWait()`

Very strange problem, anyone can help? 

Thank U very much!

kai-morich, 2020-12-15:
Do you use `SerialInputOutputManager`? There it is normal that `read` without timeout typically blocks to fetch next data. If you use `read` in your main thread, you better use timeout to return the available data.

yaphone, 2020-12-16:
I tried to set timeout like `usbSerialPort.read(response, 1000)`, but still not work, it still can't recieve data when length==32，just cause an exception. And I also tried the example of this project, not work too. And finally I use [the driver provided by original factory](http://www.wch.cn/products/CH340.html), it works well.

I still don't know why...

kai-morich, 2020-12-20:
I could not reproduce this issue with my CH34x test devices, there was no problem with length=32.
As the vendor does not provide a full data sheet, it's unclear what might be wrong.

yaphone, 2020-12-20:
Maybe, I am the only one with this problem, so  I think it is not representative. You can close this problem if If you think it is necessary.

Thank you all the same~~  💯

cofung, 2021-03-11:
I have the same issue, it block at mConnection.requestWait(), and after close the port, it receive the data from onNewData().

kai-morich, 2021-03-13:
now I could reproduce the issue.
please try `read()` with buffer size 32 instead of 8192 as used in example app.
with this size the issue was gone. 

I can only speculate about the reason:
* The CH34x uses an USB packet size of 32 bytes. If less data is available the Android/Linux USB library immediately returns the available data. If a full packet is available it tries to read the next packet. With `bulkTransfer` I got a -1 after the timeout and the data was lost. Later send data was correctly recieved.

The issue did not occur with other device types (FTDI, Prolific, ...)

yaphone, 2021-03-22:
THX@kai-morich

Finally I used the driver which provid by the original factory, I will try it again next time when I use ch34X chip.

THX all the same!

kai-morich, 2021-04-08:
for SerialInputOutputManager fixed with https://github.com/mik3y/usb-serial-for-android/commit/848d4e771398eeecbe953d716588ba3924780c88

xmutzlq, 2021-04-29:
i have the same issue,  when i send "11111111111111111111111111111111111111111111112"，the "2" is not receive，then i try to send "111111111111111111111111111111111111111111111122"，data is full receive，i do not know how to resolve，can you help me？

kai-morich, 2021-04-29:
when using SerialInputOutputManager you do not have to set the buffer size any more, as it already sets the optimal size = 32.

xmutzlq, 2021-04-30:
we have resolved yesterday，cause write data timeout is too short，we have increased timeout, finally it works well
thank you @kai-morich

closed 2021-05-02 by kai-morich