mik3y/usb-serial-for-android #335: CH340K无法使用，win测试是正常的

adsl865, 2020-12-30:
CH340C 可以使用  CH340k无法使用  不知道是什么原因 
在WIN 测试都是正常的

kai-morich, 2021-01-03:
Your issue is not clear from google translate. Please rephrase or write in english.

adsl865, 2021-01-06:
CH340C can be used, but CH340k cannot be used. I don’t know why.

kai-morich, 2021-01-06:
What is the issue? Is the device not detected at all, does connection fail or transfer fail? Have you checked this device at a Windows or Linux computer?

adsl865, 2021-01-06:
The WIN test is normal, the Raspberry Pi is also normal, and the Android test fails.

adsl865, 2021-01-06:
CH340C can be used on Android, but CH340k K suffix cannot be used.

kai-morich, 2021-01-06:
Is the device not detected at all, does connection fail or transfer fail?

adsl865, 2021-01-07:
Is the device not detected at all

kai-morich, 2021-01-07:
is the device shown with an app that lists all USB devices (e.g. [USB device info](https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator))?

closed 2021-02-05 by kai-morich