mik3y/usb-serial-for-android #336: Question about bridge to file like /data/data/<com.app.package>/fakeUSB0

araczkowski, 2021-01-04:
I'm just curious, Is it possible to use this library to interface the usb-serial directly to a file.
Of course without root, the file will be in app user space (no in /dev/). 
The idea is to do not direct read/write in Java but via file. 

Seems theoretically possible for me, but I just found this nice lib/repo and maybe I don't get it yet.
Thanks in advance for you opinion.

kai-morich, 2021-01-04:
I don't think thats possible on Android. You have to read/write with `android.hardware.usb.Usb...` classes and then can save to a file

araczkowski, 2021-01-04:
It is possible to bridge the usb-serial to tcp on Android: https://play.google.com/store/apps/details?id=com.cidtepole.serverbridge&hl=en
I would like to do something similar but with a bridge to file not to network. 
With usb-serial-for-android this task seems not so hard to achieve... but I'm completely a novice in this topic.

kai-morich, 2021-01-04:
This app looks like it's reading/writing with `android.hardware.usb.Usb... classes (or some library on top like this) and then bridging to a tcp socket. The same should be possible to a file if you write an App, but why should another app use a file instead of this library? You either would have to implement some connection property negotiation thru this file (like RFC 2217) or only use fixed baud-rate, ...

araczkowski, 2021-01-04:
I know it sound odd. But this is for other standard languages libs that we can use on Android.

Android OS does not provide a write access to system directories such as root file system ("/"). But we are able to install Linux packages in app data directory (installation prefix -> /data/data/app.package) and all works fine.

Having the possiblity to bridge the usb-serial comunication to the file in app data directory we will be able to use the standard python, nodejs, c libs to comunicate with usb devices... on Android.

araczkowski, 2021-01-04:
> But we are able to install Linux packages in app data directory (installation prefix -> /data/data/app.package) and all works fine.
I'm talking about Termux approach: https://github.com/termux https://termux.com/

kai-morich, 2021-01-06:
The /dev/.../USB files are not real files, but are device files that are typically accessed with ioctl commands. To achieve similar solution in user space I guess you would have to use a pipe file, where a Java process reads/writes the one end and translates to usb-serial-for-android commands and your python/nodejs/... app read/writes the other end of the pipe. You still would need a protocol to transfer baudrate, control lines, .... over this pipe like RFC2217 or some ioctl style. On the python/nodejs/.. side I guess nobody wants to deal with such files directly but would prefer an API similar to this library. 
To me looks more convenient to use one of the existing libraries like [usbserial4a](https://github.com/jacklinquan/usbserial4a)

araczkowski, 2021-01-06:
> The /dev/.../USB files are not real files

yes I know, these are [Character devices files](https://en.wikipedia.org/wiki/Device_file), cause [Everything is a file](https://en.wikipedia.org/wiki/Everything_is_a_file)

> To achieve similar solution in user space I guess you would have to use a pipe file...

I thought that I can just use ``mknod`` to create souch file, the same like it is done in ueventd.rc (but of course in user space not in /dev). In Termux mknod is included in BusyBox (no root access is needed).
But I didn't check this yet, still no time...

> To me looks more convenient to use one of the existing libraries like usbserial4a

Thanks for the link - I will check it :+1:

closed 2022-05-24 by kai-morich