mik3y/usb-serial-for-android #337: USB Serial android library supports flashing?

jesuelab, 2021-01-07:
I am using ATmega2560 (CH340) Microcontroller boards .Is it possible to flash by using hex file in android by using usb serial library.  I am using physicaloid library , but it does not support Ch340 . Is there any alternate libarry for flashing in android.

jesuelab, 2021-01-07:
> I am using ATmega2560 (CH340) Microcontroller boards .Is it possible to flash by using hex file in android by using usb serial library. I am using physicaloid library , but it does not support Ch340 . Is there any alternate libarry for flashing in android.

jesuelab, 2021-01-07:
I am using ATmega2560 (CH340) Microcontroller boards .Is it possible to flash by using hex file in android by using usb serial library. I am using physicaloid library , but it does not support Ch340 . Is there any alternate libarry for flashing in android.

kai-morich, 2021-01-07:
the library only provides low level serial communication. You can build a flashing library/app on top

jesuelab, 2021-01-08:
any suggestion kai ??

kai-morich, 2021-01-08:
I did not use flashing from Android, so I cannot recommend an App or Library

closed 2021-01-08 by kai-morich