mik3y/usb-serial-for-android #342: write and read data synchronously doesn't work

ptrczajka, 2021-01-22:
Hi, I am working with ESP32 (CP2102 USB to UART Bridge Controller). I need to send text command to this device and then immediately **read** response synchronously in the same function to return data. I am usingbaud rate 11520 but with lower values the same issue.  I checked also on Arduino Uno Rev 3 and the same issue.

```
private fun requestData(textMessage: String): String {
        send(textMessage)
        return read()
}

private fun send(textMessage: String) {
        val messageByteArray = "$textMessage$NEW_LINE".toByteArray()
        val writeResult = usbSerialPort.write(messageByteArray, USB_WRITE_TIMEOUT)
}
fun read(): String {
        val messageByteArray = ByteArray(1024)  //16384   multiply of 64
        val readDataLength = usbSerialPort.read(messageByteArray, USB_READ_TIMEOUT)
        return messageByteArray.decodeToString()
}

const val SERIAL_BAUD_RATE = 115200 //57600 //115200
const val VENDOR_ID = 4292 // ESP32 4292   //arduino 9025
const val PRODUCT_ID = 60000 // ESP32 60000  //arduino 67
const val USB_READ_TIMEOUT = 3000
const val USB_WRITE_TIMEOUT = 1000
const val NEW_LINE = "\n"
```

write is returning result > 0  but messageByteArray from read is empty or sometimes contains cut and messy data. 
What I'm doing wrong ? Is it possible to write and then read data in sync way ?

Code from ESP32 / Arduino

```
#include <Wire.h>

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay (100);
  }
}
 
void loop() {
  String serialANDROID = "";
  if (Serial.available() ) {
    serialANDROID = Serial.readStringUntil('\n');
    if (serialANDROID == "command1") {
      Serial.println ("someData = [123456789123456]/nEND/n/n");
    }
    else if (serialANDROID == "command2"){
      Serial.println("OK/nEND/n/n"); 
    }
    else if (serialANDROID == "command3"){
      Serial.println("isLocked=true/nEND/n/n");
    }
    else if (serialANDROID == "command4") {
      Serial.println ("someExtraData = [1234567891234568685645334354]/nEND/n/n");
    }
    else {
      Serial.println("error"); 
    }
  }
}
```

kai-morich, 2021-01-23:
* write is asynchronous: https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#Is_codewritecode_synchronous
* read can return arbitrary chunks: https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

so you should loop over `read` until the end of you message is reached or until timeout

ptrczajka, 2021-01-24:
@kai-morich Thanks for response I have improved my code and know works better.
 But I have a case when I need to cancel looping over `read` when my custom timeout is exceeded. So it looks like some unread part of data remains in buffor and next when I sending request to usb device using `write`, `read` is returning **previous data** which waiting in buffor.
 Is there any way to **clear** buffor before I will call next  `write` `read` ?

ptrczajka, 2021-01-24:
I have found potential solution.
Calling `usbSerialPort.purgeHwBuffers(true, true)` before each `write` `read` to be sure about getting last recent data without reading something old what is waiting in buffer is good way to go ?

kai-morich, 2021-01-24:
👍

closed 2021-02-07 by kai-morich