mik3y/usb-serial-for-android #343: [General Guidance] Need some help parsing USB serial packages that do not have stop bits.

Drkstr, 2021-01-25:
Hi, 
      I am trying to use this library to receive and read MAVLink messages. This is the package structure: https://mavlink.io/en/guide/serialization.html 
      
The package contains a bit that indicates the length of the package and a CRC based checksum at the end of the package. 

In your example, you have the `setParameters` method for setting the baud rate, data bits, stop bits and parity. Do you have a similar method that I can use to when the package I am trying to read does not have a stop bit?

kai-morich, 2021-01-26:
this library only provideds low level `read/write` 
protocol specific handling has to be done in your app as depicted [here](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time)

kai-morich, 2021-01-26:
btw: each serial byte is terminated by a stop bit, not a logical package

closed 2021-01-28 by Drkstr