mik3y/usb-serial-for-android #344: connection lost: Queueing USB request failed

CJH7396, 2021-01-27:
hello, I tested with many Android smartphones (with different Android versions) and it worked well, but it gives me an exception ('connection lost: Queueing USB request failed') when I test with my PDA (Android 8.1). Is there any reason why it keeps giving me above exception?

Manicben, 2021-02-02:
Hi, I'd just like to add that I'm also observing this issue when calling `port.read(buffer, 0)` with buffer size of 1.

I'm on Android 11 and I've tested with an Arduino Uno (CDC driver), Arduino Nano (CH340 driver) and Sparkfun Pro Micro (ATMega32u4 with CDC driver), all give the same IOException of `Queueing USB request failed`. I've tried ignoring the exception and retrying forever, but to no avail. I know for certain that the Arduinos are sending data to the Android phone.

I am able to `port.write(buffer, 0)` with no issues on the other hand and I can see the Arduinos have indeed received the written bytes.

CJH7396, 2021-02-03:
> Hi, I'd just like to add that I'm also observing this issue when calling `port.read(buffer, 0)` with buffer size of 1.
> 
> I'm on Android 11 and I've tested with an Arduino Uno (CDC driver), Arduino Nano (CH340 driver) and Sparkfun Pro Micro (ATMega32u4 with CDC driver), all give the same IOException of `Queueing USB request failed`. I've tried ignoring the exception and retrying forever, but to no avail. I know for certain that the Arduinos are sending data to the Android phone and the correct baud rate is set on both sides, 9600.
> 
> I am able to `port.write(buffer, 0)` with no issues on the other hand and I can see the Arduinos have indeed received the written bytes.

closed 2021-02-03 by CJH7396