mik3y/usb-serial-for-android #345: please support a special Prolific chip:  VID_067B&PID_23C3

EiChinn, 2021-01-29:
Hi,
this library works perfectly on a cable with the Prolific chip:  VID_067B&PID_2303, thank you very much!
But now I have [the same cable product](https://www.lulian.cn/product/142-cn.html) with [a new Prolific chip](http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=225&pcid=41):  VID_067B&PID_23C3. When I use this new cable, it can't be recognized by the library. Then I add the PID 0x23C3 to the "UsbId.java" and "ProlificSerialDriver.java" file. Now it can be successfully recognized but connection failed with an error message: "ControlTransfer 0x8 failed: -1". This happens when the method "purgeHwBuffers()"(in ProlificSerialDriver.java) is called.
 I tried to comment out the implementation of these methods: resetDevice(), doBlackMagic() in ProlificSerialDriver.java.  Connection Succeed and it seems to be able to communicate with external devices normally. But I don't know what side effects such changes will have.
So, could you please support this Prolific chip(VID_067B&PID_23C3)?

kai-morich, 2021-01-30:
This is a completely different chip with similar looking name. It needs a completely different driver as mentioned [here](https://www.mail-archive.com/linux-usb@vger.kernel.org/msg112041.html) for Linux. As i havent found the new chips anywhere for reasonabl prices, its difficult to add & test the new driver variant.

zhuqiyun, 2021-03-27:
> This is a completely different chip with similar looking name. It needs a completely different driver as mentioned [here](https://www.mail-archive.com/linux-usb@vger.kernel.org/msg112041.html) for [...]
when to support ，thx

bluetoblack9, 2021-04-19:
Hi,
Looking forward to your support for VID_067B&PID_23C3.

irregular2015, 2021-04-29:
Hi. If you are looking into this, have you encounter the problem that its reading modern status(CTS, DCD, DSR, RI) are always ON?
I was using the SDK from its official site: http://www.prolific.com.tw/UserFiles/files/PL2303G_Android_v1000_20180827.zip
Look forward to your support.

irregular2015, 2021-05-13:
> Hi. If you are looking into this, have you encounter the problem that its reading modern status(CTS, DCD, DSR, RI) are always ON?
> I was using the SDK from its official site: http://www.prolific.com.tw/UserFiles/files/PL2303G_Android_v1000_20180827.zip
> Look forward to your support.

Finally, I find out the controlTranfer() command inside PL2303G_GetCommModemStatus(), the third args is wrong, not 0x87, it is 0x8080!! Hope it can help you.

kai-morich, 2021-05-13:
Im currently working onPL2030G support and also noticed that 0x87 does not work. Looks like the registers repeat each 0x20, so all these worked: 0x00, 0x80 and 0x8080. The bits are also at other positions.

kai-morich, 2021-05-14:
https://github.com/mik3y/usb-serial-for-android/commit/c82cd284ae45f6a86169be3c46fff943a9c80ad8

closed 2021-05-14 by kai-morich