mik3y/usb-serial-for-android #346: Direct Read with infinite timeout causing IOException: Queueing USB request failed

Manicben, 2021-02-03:
Hi there, when performing a direct read with infinite timeout by calling `port.read(buffer, 0)` with buffer size of 1, `IOException: Queueing USB request failed` is always raised, no matter what I do. 

I'm on Android 11 and I've tested with an Arduino Uno (CDC driver), Arduino Nano (CH340 driver) and Sparkfun Pro Micro (ATMega32u4 with CDC driver), all give the same IOException . I've tried ignoring the exception and retrying forever, but to no avail. I know for certain that the Arduinos are sending data to the Android phone and the correct baud rate is set on both sides, 9600.

On the other hand, I am able to perform a direct write using `port.write(buffer, 0)` with no issues and I can see the Arduinos have indeed received the written bytes.

kai-morich, 2021-02-03:
If more than one byte is send, you might get this error as mentioned [here](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#How_to_size_the_codereadcode_buffer). please try with buffer size 64.

Manicben, 2021-02-03:
Hey there, thanks for the quick response!

I've now tried with `byte[] buffer = new byte[64];` instead of `byte[] buffer = new byte[1];`, but this made no difference.
I still get `java.io.IOException: Queueing USB request failed` thrown by `port.read(buffer, 0)`.

If I try specifying a timeout, like 5000ms, I get `java.io.IOException: USB get_status request failed` from `port.read(buffer, 5000)`.

kai-morich, 2021-02-05:
does it work with the sample app included in this repo?

Manicben, 2021-02-06:
It sure did, and I figured out my issues. One thing to note is that the application code I'm working with wasn't written by me and was previously using the Physicaloid library, which buffers read bytes internally and then serves that through its API.

- I switched to using timeouts, made sense to for my use case anyway (but I think setting infinite timeouts would also work)
- The application I was refactoring was expecting to be able to get back 1 byte at a time, this of course doesn't work here as all bytes must be written to the provided buffer at once, rather than providing a buffer of size 1 and internally buffering the rest as per Physicaloid's behaviour.
- Once I refactored the application to instead iterate over the returned buffer instead of fetching 1 byte at a time, everything started working.

Now I only get those IOExceptions when the USB serial device is unplugged whilst the port is open (which is expected).

Many thanks, the example app proved that the application code was at fault, not the library, and that helped me narrow down what was wrong in the application!

closed 2021-02-06 by Manicben