mik3y/usb-serial-for-android #347: 3.3.0 does not built without forcing java version to 1.8

janimm, 2021-02-12:
3.3.0 causes build error:

    Invoke-customs are only supported starting with Android O (--min-api 26)

If can be fixed by adding following to the build.gradle:

    android {
        compileOptions {
            sourceCompatibility JavaVersion.VERSION_1_8
            targetCompatibility JavaVersion.VERSION_1_8
        }
    }

3.2.0 did not have this issue.

kai-morich, 2021-02-12:
java 1.8 is already set since 3.0.0 with https://github.com/mik3y/usb-serial-for-android/commit/a664082f23e680bbf847ec6bfef67296139f289c
is it missing in another gradle file?

janimm, 2021-02-15:
I'm not a gradle expert, but this is what I'm experiencing while building our project:

In our build.gradle we have:
    dependencies {
        ...
        implementation 'com.github.mik3y:usb-serial-for-android:3.2.0'
    }

If I change that to:
    dependencies {
        ...
        implementation 'com.github.mik3y:usb-serial-for-android:3.3.0'
    }

The build fails with error:
    Invoke-customs are only supported starting with Android O (--min-api 26)

Which can be fixed by adding this to build.gradle:
    android {
        compileOptions {
            sourceCompatibility JavaVersion.VERSION_1_8
            targetCompatibility JavaVersion.VERSION_1_8
        }
    }

I know this is not a bug and fix is easy, but something has happened between 3.2.0 and 3.3.0 to require this change into the build.grade.

kai-morich, 2021-02-16:
I see. Google mentions [here](https://developer.android.com/studio/write/java8-support):

> After that, for each module that uses Java 8 language features (either in its source code or **through dependencies**), update the module's build.gradle file

so any app using this library also has to use Java 8. With a commit between 3.2.0 and 3.3.0 a Java 8 feature (method reference) is used, so this option becomes effective now.

I can mention this in the wiki page, but searching for this error message also quickly shows the resolution.

closed 2021-11-02 by janimm