mik3y/usb-serial-for-android #352: Ошибка при чтении порта FTDI

NetFix1, 2021-03-17:
In case only 2 bytes are read, the readFilter function and then the read function return 0.

line 167 FtdiSerialDriver.java

`int length = Math.min(srcPos + maxPacketSize, totalBytesRead) - (srcPos + READ_HEADER_LENGTH);`

NetFix1, 2021-03-17:
The error does not appear on version 2.3.0

kai-morich, 2021-03-19:
the FTDI response starts with 2 header bytes. The code was different in the past but also stripped this header:
https://github.com/mik3y/usb-serial-for-android/blob/a1e58b984389d6310c81d81c4f7f8ad7e9c4ea3a/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java#L239

NetFix1, 2021-03-20:
I look at the trace with the debugger, I see my 2 bytes coming from my device and how the read function returns: 0 bytes read ...

NetFix1, 2021-03-20:
I also tested adapters ch430, cp2102 and pl2303, they work fine on the latest version of the library.
Until the problem was solved by downgrading to 2.3.0, FTDI is working with it.

P.S. Sorry for google translation

kai-morich, 2021-03-20:
FTDI devices send a an empty response each 16msec that contains the 2 header bytes only, so your debugger probably stopped at this _empty_ response. To debug your response with X bytes payload, you should look for a response with X+2 bytes in total.

NetFix1, 2021-03-20:
`private int readFilter(byte[] buffer, int totalBytesRead) throws IOException {
            final int maxPacketSize = mReadEndpoint.getMaxPacketSize();
            int destPos = 0;
            for(int srcPos = 0; srcPos < totalBytesRead; srcPos += maxPacketSize) {
                int length = Math.min(srcPos + maxPacketSize, totalBytesRead) - (srcPos + READ_HEADER_LENGTH);
                if (length < 0)
                    throw new IOException("Expected at least " + READ_HEADER_LENGTH + " bytes");
                System.arraycopy(buffer, srcPos + READ_HEADER_LENGTH, buffer, destPos, length);
                destPos += length;
            }
            //Log.d(TAG, "read filter " + totalBytesRead + " -> " + destPos);
            return destPos;
        }`
The contents of the destPos variable after the execution of the function = 0;

kai-morich, 2021-03-20:
I created a mockito/robolectric unit test for `readFilter`, that I will check in soon. It behaves identical for 2.3.0 and 3.x

    input length -> output length
    2 -> 0
    3 -> 1
    4 -> 2
    ...
    64 -> 62
    66 -> 62
    67 -> 63
    n*64 -> n*62

I asume you now observe different behavior from `read` as it now has improved retry + timeout handling and does not return any more with empty response after status packet is received each 16msec.

kai-morich, 2021-04-16:
unit test: https://github.com/mik3y/usb-serial-for-android/blob/848d4e771398eeecbe953d716588ba3924780c88/usbSerialForAndroid/src/test/java/com/hoho/android/usbserial/driver/FtdiSerialDriverTest.java#L38

closed 2021-04-16 by kai-morich

NetFix1, 2021-04-17:
The fact is that all other drivers work, but this one does not ...

NetFix1, 2021-04-17:
And everything works fine on the younger version of the library.

kai-morich, 2021-04-17:
When you get 2 bytes from the device, this is only the FTDI header.
Did you get 4 bytes from the device containing header + 2 bytes data?

NetFix1, 2021-04-17:
I need to choose how to process data from the device depending on the adapter? 

I do not receive data from FTDI at all

NetFix1, 2021-04-17:
By the way, the library is excellent, I am very grateful to you for its development.

kai-morich, 2021-04-17:
should work without adapter specific coding

optisamit, 2023-10-06:
Привет @NetFix1 
Хотел бы тебя спросить кое о чем касательно протокола. Ты мог бы написать мне в телеге? @optisamit