mik3y/usb-serial-for-android #353: [Question] NoClassDefFoundError Occured...

pegaskim, 2021-03-29:
I am using the usb to serial library usefully.
Thank you for making a good library.

I get the following error while using, but I don't know where to find help, so I'm posting it here.

After separating the class that uses usb-serial-for-android:3.2.0 in the app project into aar library that works without any problems, the following error occurs when running.
  Process: com.gen2wave.smartsledtestapp, PID: 20552
     java.lang.NoClassDefFoundError: Failed resolution of: Lcom/hoho/android/usbserial/driver/ProbeTable;

This is an error that occurs in the code after declaration to use a custom USB.
ProbeTable customTable = new ProbeTable();
(For reference, it works well without any problems, including the above code as Class, not aar.)

Can I get any help in connection with it?

kai-morich, 2021-03-29:
maybe you have to tell proguard to keep some code like used [here](https://github.com/mik3y/usb-serial-for-android/blob/b6e9dbe40f297c7cbdd5f7e24297c867f6fb8eb0/usbSerialForAndroid/proguard-rules.pro#L1)

pegaskim, 2021-03-29:
i already set.. but, error.
my proguard-rules.pro set below line
-keep class com.hoho.android.usbserial.driver.* { *; }

and build.gradle file is below.
and build release version module.

/// build.gradle start
buildscript {
    repositories {
        // append for com.github.mik3y dependencies
        maven { url 'https://www.jitpack.io' }
    }
    dependencies {
        classpath "com.github.mik3y:usb-serial-for-android:3.2.0"
    }
}

plugins {
    id 'com.android.library'
}

android {
    compileSdkVersion 29
    buildToolsVersion "29.0.3"

    defaultConfig {
        minSdkVersion 26
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        multiDexEnabled true

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
        consumerProguardFiles "consumer-rules.pro"
    }

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }

        debug {
            minifyEnabled false
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}

dependencies {
    implementation 'androidx.appcompat:appcompat:1.2.0'
    implementation 'com.google.android.material:material:1.3.0'
    testImplementation 'junit:junit:4.13'
    androidTestImplementation 'androidx.test.ext:junit:1.1.2'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.3.0'
    implementation 'androidx.multidex:multidex:2.0.1'
    implementation 'com.github.mik3y:usb-serial-for-android:3.2.0'
}
/// build.gradle end.

pegaskim, 2021-04-07:
Instead of using usb-serial-for-android aar in my library source, it was solved by build including usb-serial-for-android source.

closed 2021-04-07 by pegaskim