mik3y/usb-serial-for-android #355: Support pid_067b vid_2304

NetFix1, 2021-04-08:
Users found somewhere an adapter USB - RS232(PL2304), is it possible to add its support?

kai-morich, 2021-04-08:
Please try with custom prober mapping 2304 to the existing prolific driver. If that works the additional VID could be added to the library.

gdudek, 2022-02-20:
It's already well documented by Kai, but the exact  code you need may look like (in Kotlin):
`
            val customTable = ProbeTable()
            customTable.addProduct(0x2304, ox067b, CdcAcmSerialDriver::class.java)
            val prober = UsbSerialProber(customTable)
            availableDrivers = prober.findAllDrivers(manager)
            driver = availableDrivers[0]
`