mik3y/usb-serial-for-android #361: Galaxy S20 to Arduino Nano

tesorrells, 2021-05-05:
I am trying to send some serial data from my android app to an arduino nano, just some numbers, and when i hit the button to send the nano RX light flashes, but the code on the nano doesn't do anything, as it should, with the data. I'm using a USB-C to USB-C cable, as when i tried to usb an OTG cable it did not recognize the USB device was even there on the android side. Not sure if the issue is on the android side or the nano side. 

`protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        cameraButton = findViewById(R.id.cameraButton);
        detectButton = findViewById(R.id.detectButton);
        imageView = findViewById(R.id.imageView);

        // usb connection
        // Find all available drivers from attached devices.
        UsbManager manager = (UsbManager) getSystemService(Context.USB_SERVICE);
        List<UsbSerialDriver> availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(manager);
        PendingIntent mPermissionIntent = PendingIntent.getBroadcast(this, 0, new Intent(
                ACTION_USB_PERMISSION), 0);
        IntentFilter filter = new IntentFilter(ACTION_USB_PERMISSION);
        registerReceiver(usbReceiver, filter);
        if (availableDrivers.isEmpty()) {
            this.sourceBitmap = Utils.getBitmapFromAsset(MainActivity.this, "muj.jpg");

            this.cropBitmap = Utils.processBitmap(sourceBitmap, TF_OD_API_INPUT_SIZE);

            this.imageView.setImageBitmap(cropBitmap);
            return;
        }

        // Open a connection to the first available driver.
        UsbSerialDriver driver = availableDrivers.get(0);
        UsbDeviceConnection connection = manager.openDevice(driver.getDevice());

        UsbDevice device = driver.getDevice();
        if (connection != null) {
            manager.requestPermission(device, mPermissionIntent);
        }

        UsbSerialPort port = driver.getPorts().get(0); // Most devices have just one port (port 0)
        try {
            port.open(connection);
        } catch (IOException e) {
            e.printStackTrace();
        }
        try {
            port.setParameters(115200, 8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);
        } catch (IOException e) {
            e.printStackTrace();
        }

        cameraButton.setOnClickListener(v ->
                startActivity(new Intent(MainActivity.this, DetectorActivity.class)));

        detectButton.setOnClickListener(v -> {

            Handler handler = new Handler();

            new Thread(() -> {
                handler.post(new Runnable() {
                    @Override
                    public void run() {
                        try {
                            port.write("60".getBytes(), 5);
                        } catch (IOException e) {
                            e.printStackTrace();
                        }
                        byte response[] = {};
                        try {
                            port.read(response, 5);
                        } catch (IOException e) {
                            e.printStackTrace();
                        }
                    }
                });
            }).start();

        });

        initBox();
    }
`

kai-morich, 2021-05-06:
typical issues are wrong baud rate or incomplete permission handling. please try with example app.

vish897, 2021-05-23:
> typical issues are wrong baud rate or incomplete permission handling. please try with example app.

Hey! I am having the same issue. I am trying to stream the data continuosly. I tried sending a string from android app to Arduino Uno, I can see the Rx led glowing but LED is not blinling as it should according to the program.
However I observed that when I am sending the string just once from the app, the LED blinks but it takes some time to do this after the Rx pin blinks. I think this is the reason for the stream of data working. Is there a way I could set frequency of streaming?

closed 2021-07-21 by kai-morich